makerbase-mks/MKS-Robin #110: Mks robin mini v2.0 does not update. Display error?

RENSO1997, 2019-07-21:
Hi, I'm very angry with the people of MKS. Buy the mks robin mini v2.0. I try to update the configuration file, but the only thing that updates is the appearance of the screen. Mark an error message with red letters at the top right. It does not heat the extruder, you can not change the current value of the motors. There is no way to upgrade to version 2.0.8. I need urgent solutions from the people of mks.
[image]

makerbase-mks, 2019-07-22:
1、Display error, please ensure that the extruder heat and hot bed thermal connection is correct.
2、If you need to update the firmware, please put the four files “robin_mini.bin、robin_mini_config.txt、mks_pic、mks_font“ on the sd card (the SD card is formatted as fat32) to update.

shivahegde, 2019-10-15:
Hi RENSO1997,
Did you solve this problem? please let me know if yes.

L0rdHighlander, 2024-03-05:
Hello, I know it's not related but since you connected the endstops to the mks robin mini I tried to do it but they don't work