makerbase-mks/MKS-Robin #112: How to upload Marlin firmware in mks robin stm32

github051, 2019-08-02:


Filcanna, 2019-08-03:
I think you must put compiled firmware (within .pioenvs folder if using Platform.io) on board SD card and reset. That's it.

tdragovich, 2019-08-08:
I have been working on this on and off for a month, but as far as I can tell the procedure wants to access the sd card via the board. So you have to have access to the board in order to compile, and frankly I my board is attached to the printer inside an enclosure and frankly i am to lazy to pull it out to try something that may not work. 

With that said, per the walk-through for arm boards, we have to select the port (with the board attached via usb). I reminds me of using arduino ide to select the board, from there you would select the chip ( also like arduino), then compile and it will send it to the SD Card inside the board. Disconnect and restart board.... 

If someone else has had success please help us out here.

3Dangel, 2019-08-08:
No.
You do not need to access tour board to compile FW.
You do it at your PC.
32stm’s FW is compiles by using the Atom-platformio

Check YouTube
https://youtu.be/gMPdsBumWCQ

The process is clear on how you should be doing , just change the board
used ...etc
Good luck

github051, 2019-10-29:
Thanks for help. I got it now.

closed 2019-10-29 by github051

Filcanna, 2019-10-30:
Open source Marlin firmware is available on MKS site (I think they at last follow pressing questions by the 3D community, asking for opening their fw). Also can be found original fw, allowing to customize printer settings by a text file on the SD card directory root. I verified that fw is seen by serial monitor on Arduino, so I think card could also be driven by Octoprint