makerbase-mks/MKS-Robin #113: Marlin on MKS Robin Nano

artdezigner, 2019-08-02:
Good day, please tell me how to configure Marlin. As I understand it, I need to migrate the basic settings from robin_nano_cfg.txt in corfigure.h, the board and the display you have already registered, but I can not understand what numbers to set in variables SERIAL_PORT and what BAUDRATE speed to specify: 
/**
* Select the serial port on the Board that will be used to communicate with the host.
* This allows you to connect wireless adapters (for example) to non-standard port pins.
* Note: the first serial port (-1 or 0) will always be used with the Arduino bootloader.
* :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
*/
#define SERIAL_PORT **3**

/**
* Select the additional serial port on the Board that will be used to communicate with the host.
* This allows you to connect wireless adapters (for example) to non-standard port pins.
* Serial port -1 is an emulated USB serial port, if available.
* :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
*/
#define SERIAL_PORT_2 **1**
#define NUM_SERIAL **2**

/**
* This setting determines the printer's data rate.
*/
#define BAUDRATE **250000**

makerbase-mks, 2019-10-11:
1, the baud rate can be set according to the general;
2, serial port
A: 32-bit CPU board, namely MKS Robin series, MKS SBASE/SGEN/SGEN_L and so on, need to be set according to which group of serial ports are actually connected. MKS has open sourced the above schematic diagram of the motherboard, you can refer to it;
B: MKS Robin Nano or other MKS 32 motherboard, #define SERIAL_PORT -1, here must be -1, this macro is used in 2560 and other 8-bit CPU;
C: Because the USB of MKS Robin Nano is connected to STM32F103VET6 PB10/PB11, this is USART3, so #define SERIAL_PORT_2 3

daniellezama, 2025-03-06:
Hi i have the same problem with a Mks robin tft 3.5 v1 with a mks robin nano v1.3 board  
In the manual says #define SERIAL_PORT 3 only and not works.