makerbase-mks/MKS-Robin #114: NO PIN SCHEMATICS for Robin lite and lite2

Filcanna, 2019-08-03:
I didn't succeed to find pin description/number for those boards, either on corresponding folder (where is also marlin firmware as of version 2 bugfix) or within datasheet folder of Makerbase github site. I'm looking for a board seen from top with clear pin numbers on it, not for description of connections with hotend, motors, thermistors and so on...

makerbase-mks, 2019-10-11:
MKS Robin Lite and Lite2 have been open source SCH, you can refer to
MKS Robin Lite: https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin%20Lite
MKS Robin Lite2: https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin%20Lite2

rofoto, 2021-08-05:
this folder does not contain the schematics for the robin lite v1 or v1.1 only the binary blobs for the firmware 
此文件夹不包含 robin lite v1 或 v1.1 的原理图，仅包含固件的二进制 blob

this is what the MKS ROBIN folder looks like
这就是 MKS ROBIN 文件夹的样子 
[image]

and this is the MKS ROBIN LITE folder 
这是 MKS ROBIN LITE 文件夹 
[image]
如您所见，缺少 HARDWARE 文件夹

is there any chance to get this updated?
有没有机会得到这个更新？

ZeroedByte, 2022-10-25:
Bump. Having the same issue, trying to troubleshoot a MKS Robin Lite and am having no luck since there's no schematic.