makerbase-mks/MKS-Robin #115: Is it an error on platformio.ini for Robin Lite/LIte2

Filcanna, 2019-08-03:
[image]

Can it be an error? Only God knows ;)

3Dangel, 2019-08-03:
Hello Filcana,
Please go to YouTube and search for mks-sbase Marlin
And there will be tutorials ( step by step) and you will understand.
Good luck

Filcanna, 2019-08-03:
Thanks. But in reality I’ve already compiled Marlin 2 bug fix for my Robin Lite board.   Only I think is right to have a right and univocal documentation.

makerbase-mks, 2019-10-11:
1、You cna change it
2、You can use the PIO Build in the lower left corner of the ATOM page, then search for MKS Robin Lite in the dialog box and select PIO Build (mks robin lite).