makerbase-mks/MKS-Robin #116: Hi makerbase, is there any PWM free pins are available in MKS Robin stm32 ?

github051, 2019-08-05:


jfkerekes, 2019-08-14:
I am wondering the same thing for the Robin MINI... want to know if pin PB2 can be used for BLTouch control.

zillarob, 2019-08-28:
Look at the data sheet for the chip on your board to see what pins do what, then look at the schematic to figure what pins are free, or can be freed up.

makerbase-mks, 2019-10-11:
MKS Robin hardware is open source, you can refer to: https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin/Hardware

3Dangel, 2019-10-11:
Glad to see MKS is responding to the questions. Pls keep it coming.

My MKs-Robin v2.1 with WIFI module can not get the Marlin 2 to work.
It complies correctly.
But no LCD screen after FW being flashed in.
It stays on updating.
Even after pressing reset button
It restart but the lcd only shows “booting”.
The version on mks-GitHub of Marlin is old and does not support mixing dual
color.
 Would you be able to test at your end? and find out what is wrong and what
are the fixes?

Thanks in advance for your help

Robert

makerbase-mks, 2019-10-14:
Currently, Marlin 2.0 firmware does not support MKS Robin WIFI.