makerbase-mks/MKS-Robin #118: MKS Robin Mini with Marlin 2.0 Extruder does not feed

jfkerekes, 2019-08-09:
I am having an issue with Marlin 2.0 on an MKS Robin Mini. Got it compiled and everything works as expected except that the extruder stepper will not turn. No grinding or whining indicating low stepper current, just will not turn. Can't get it to move through pronterface or Octoprint control interface and will not extrude during printing. Hot End heats up all other axis's work as expected... just no extruder. It is like that axis does not exist. If I flash back to the MKS firmware I can print normally, no problem so I know it isn't a hardware issue. Has ANYONE else had this issue? My printer support guy has tested on the same model printer at the factory with the same Robin Mini board and got the same result. Any help would be appreciated. I have looked through Configuration.h, configuration_adv.h and the pins file for the Robin Mini and can find nothing that stands out as being the issue.

jfkerekes, 2019-08-14:
RESOLVED - This was a pin conflict. Corrected and working.

closed 2019-08-14 by jfkerekes

3Dangel, 2019-08-14:
If pin conflict: was it came from the source. ( github)?
If so, please post the before and after so the Marlin’s Dev. Team can make
the changes and others can be benefit from your finding.

Cheers

Mectrotech, 2020-02-16:
Hi ,
i have the same problem with board MKS robin V 2.0 
Someone can help me?
thanks

makerbase-mks, 2020-02-17:
@Mectrotech It is also a pin conflict?

makerbase-mks, 2020-02-17:
If E1 AXIS not work; you can refter to:
#define E1_ENABLE_PIN PA12
#define E1_STEP_PIN PA11
#define E1_DIR_PIN PA8
Direction signal PA8 is no problem;
The enable signal and the pulse signal are actually USB port function pins, so in the configuration.h file, SERIAL_PORT cannot be defined as -1, otherwise the PA12 and PA11 pins will be disabled.So change it to 1, PA12, PA11 pins will not be disabled