makerbase-mks/MKS-Robin #119: Conexión del doble extrusor a la placa MKS Robin nano

jecopo6, 2019-08-09:
He montado el doble extrusor en mi impresora Tevo Nereus. Tengo conectados los termistores y funcionan bien midiendo la temperatura de los dos extrusores.

 Pero el problema lo tengo al conectar los cartuchos calentadores. 

El cartucho del primer extrusor va conectado en (HE0) y el ventilador fijo de los exteriores en (HE1), por lo que cuando conecto el primer extrusor (HE0) directamente se acciona el ventilador (HE1).

Con lo cual si conecto el cartucho calefactor del segundo extrusor en (HE1) no me regula la temperatura, el termistor da la lectura del extrusor pero al calefactor no para de llegarle tensión como si fuera un ventilador.

Cómo debería conectar el extrusor? Cómo debo configurar la programación para que me regule el segundo extrusor?

3Dangel, 2019-08-09:
Pie so we deberías utilizar inglés para ip tener mejor ayuda del mundo.
No has especificado que FW estás utilizando. Pero como todos tipos the FW,
lis exteusores y su respectivo calentador se tienen que asignar/ activar en
FW.
Suerte

jecopo6, 2019-08-10:
I have mounted the double extruder on my Tevo Nereus printer. I have the thermistors connected and they work well by measuring the temperature of the two extruders.

But I have the problem connecting the heating cartridges.

The cartridge of the first extruder is connected in (HE0) and the fixed outdoor fan in (HE1), so when I connect the first extruder (HE0) directly the fan (HE1) is activated.

So if I connect the heating cartridge of the second extruder in (HE1) the temperature does not regulate me, the thermistor gives the reading of the extruder but the heater does not stop reaching tension as if it were a fan.

How should I connect the extruder? How should I configure the programming to be regulated by the second extruder?

jecopo6, 2019-08-10:
Gracias por tu respuesta.

El FW que estoy usando es el 1.1

Espero que alguien me pueda ayudar porque soy nuevo en esta materia

jecopo6, 2019-08-10:
Thanks for your answer.

The FW that I am using is 1.1

I hope someone can help me because I am new in this area

jecopo6, 2019-08-10:
#===================Advanced  Settings=======================================
>cfg_insert_det_module			0			#Assemble power outage module (1:mks 220det;0:mks pwc)

>cfg_have_ups_device			0			# UPS(1:yes;0:no)

>cfg_print_over_auto_close		1			# auto-off when print finishes(1:yes£»0:no)

>cfg_filament_det0_trigger_level	0			#the level signal of  outage detection E0 module (1:high level;0:low level)
>cfg_filament_det1_trigger_level	0			#the level signal of  outage detection E1 module (1:high level;0:low level)

>cfg_filament_load_length		100			#the lenght to extrude filament (mm),Max:2000mm
>cfg_filament_load_speed		800			#the speed to extrude filament(mm/min)
>cfg_filament_load_limit_temperature	200			#It is the minimum temperature to extrude filament .

>cfg_filament_unload_length		100			#It is the minimum temperature to extrude filament .
>cfg_filament_unload_speed		800			#the speed to retract filament(mm/min)
>cfg_filament_unload_limit_temperature	200			#It is the minimum temperature to retract filament .

#=========================== Bed Leveling  =============================
#Leveling mode settings
>cfg_leveling_mode			1			#1:auto-leveling; 0:manual leveling

#=========================== Manual Leveling ==================================

>cfg_point_number			5			#the point number of manual leveling(3,4,5 point available)

#the 5 point location of  manual leveling
>cfg_point1:50,50						
>cfg_point2:180,50
>cfg_point3:180,180						
>cfg_point4:50,180
>cfg_point5:100,100

#========================== Auto Leveling ===================================

>cfg_auto_leveling_cmd:G28;G29;                                #the setting of auto-leveling command button

#=================== Language settings ==========================================
>cfg_language_adjust_type		1	#multi-language(enable:1, disable:0)

>cfg_language_type			2	#languages setting,this configuration is valid when "cfg_multiple_language" is disabled. 
						#(simplified Chinese:1; traditional Chinese:2; English:3; Russian:4; Spanish:5;French:6;Italian:7).
#==================Display Effect(refer to MKS TFT datasheet)=================================
#user-defined function1
>setmenu_func1_display		1				#1:display this button£¬0:no

#each command must be separated by semicolon ";"
>setmenu_func1:M84;

>moreitem_pic_cnt		0				#the number of "more" button 

#edit command for 1~7 "More" button
#each command must be separated by semicolon ";"
#each icon command must be less than 200 bytes
>moreitem_button1_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button2_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button3_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button4_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button5_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button6_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button7_cmd:G28 X0;G28 Y0;G28 Z0;

#the number of "more" button to dispaly on printing interface
>morefunc_cnt			0
[... 194 lines cut ...]
>BACK_PROBE_BED_POSITION 	180

>MESH_INSET			20	# Mesh inset margin on print area for MESH_BED_LEVELING

#============================== Delta Settings =============================
>DELTA_SEGMENTS_PER_SECOND	40      #--default
>DELTA_DIAGONAL_ROD		346.75	#Center-to-center distance of the holes in the diagonal push rods.	
>DELTA_SMOOTH_ROD_OFFSET	211.5	#Horizontal offset from middle of printer to smooth rod center.
>DELTA_EFFECTOR_OFFSET		28	#Horizontal offset of the universal joints on the end effector.
>DELTA_CARRIAGE_OFFSET		14.5	#Horizontal offset of the universal joints on the carriages.
>DELTA_RADIUS			169	#Horizontal distance bridged by diagonal push rods when effector is centered.	
>DELTA_HEIGHT			302	#height from z=0.00 to home position
>DELTA_PRINTABLE_RADIUS		125	#Print surface diameter/2 minus unreachable space (avoid collisions with vertical towers).
>DELTA_CALIBRATION_RADIUS	100	#set the radius for the calibration probe points - max 0.8 * DELTA_PRINTABLE_RADIUS 

#============================== Wifi Settings =============================
>CFG_WIFI_MODE			0			#wifi mode(0:sta;1:ap)
>CFG_WIFI_AP_NAME		WiFitest		#wifi name
>CFG_WIFI_KEY_CODE		makerbase		#wifi password

>CFG_CLOUD_ENABLE		1			#cloud service enable(0:disable;  1:enable)
>CFG_WIFI_CLOUD_HOST		www.baizhongyun.cn	#cloud server url
>CFG_CLOUD_PORT			10086			#cloud server port

#end.

zillarob, 2019-08-29:
Sounds like E0 autofan on E1 heater is screwing with it.
Some versions of the mks precompiled fw had it enabled, some didnt.
I didnt think the ones here had that enabled. I made the mistake of playing with them when I 1st got my Nereus and played hell trying to find one that had the autofan enabled again.
I was only single ext so you would want one that didnt have the autofan enabled.
The joys of a locked down fw =D