makerbase-mks/MKS-Robin #120: MKS-Robin2 Marlin 2.0 firmware compile not successful

DaVincIan, 2019-08-13:
I cannot get the MKS-Robin2 Marling 2.0 firmware to compile.

The first error I get is as follows:
Compiling .pioenvs\mks_robin2\FrameworkArduino\stm32\clock.c.o
In file included from C:\users\ian\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/pinmap.h:22:0,
                 from C:\users\ian\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PeripheralPins.h:34,
                 from C:\users\ian\.platformio\packages\framework-arduinoststm32\variants\MKS_Robin2/variant.h:37,
                 from C:\users\ian\.platformio\packages\framework-arduinoststm32\system\STM32F4xx/stm32f4xx_hal_conf.h:4,
                 from C:\users\ian\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\In
c/stm32f4xx_hal.h:30,
                 from C:\users\ian\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F4
xx\Include/stm32f4xx.h:250,
                 from C:\users\ian\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\stm32_def.h:35,
                 from C:\users\ian\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\PortNames.h:33,
                 from C:\users\ian\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\PortNames.c:30:
**C:\users\ian\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:16:12: error: 'PortA' undecla
red here (not in a function)**
   PA_0  = (PortA << 4) + 0x00,
            ^~~~~

I'm using PlatformIO
I managed to compile the "Robin" firmware version.
Source code was down loaded form "MKS-Robin/MKS Robin2/firmware/Marlin2.0-MKS-Robin2/"

DaVincIan, 2019-08-20:
It seems the problem lies with the newer version of ststm32 (5.5.0) see issue: platformio/platform-ststm32#267

I modified in PlatformIO.ini
[env:mks_robin2]
platform = ststm32@5.4.3

closed 2019-08-20 by DaVincIan

Evg33, 2019-08-29:
This does not solve the problem. This is a crutch ....