makerbase-mks/MKS-Robin #121: Complete files for Robin nano with 3.5 display

alfrankgit, 2019-08-24:
Please release the complete file set for the Robin nano with 3.5" display as used in the Two Trees Sapphire Pro printer, etc.

The release 1.0.2 doesn't work on this display, just a blank screen.

I suspect the pic files are wrong.

ph1lloops, 2019-08-28:
Same problem here. Did you find a solution to make it work or another way to use the LCD again? Getting the white screen without any graphics.

alfrankgit, 2019-08-29:
Not from MKS, no, why should they help so soon? :-(

Two Trees finally sent me a new firmware, but the old pics don’t completely match.

ph1lloops, 2019-08-29:
Nice could you send me that firmware please?:-)

LG Philipp

alfrankgit, 2019-08-29:
It isn’t officially released because not fully finished yet. So, no, sorry, not yet.

pcopa, 2019-10-01:
Have the same issue after my first update. After trying to change the SSID my screen went blank.

makerbase-mks, 2019-10-31:
MKS Robin Nano motherboard software already supports MKS Robin TFT35 display.