makerbase-mks/MKS-Robin #122: Marlin2.0-MKS-Robin2 compiler error

cjsoong, 2019-08-28:
Compiling with atom or VScode will result in the following error.
[image]

DaVincIan, 2019-08-28:
See issue: It seems the problem lies with the newer version of ststm32 (5.5.0) see issue: platformio/platform-ststm32#267

And closed issue #120

I have been struggling to compile Marlin2.0 MKS Robin2 with TFT option.
The best success I get is using the TFT declaration in the official Marlin 2.0 Configuration.h file. 
_**//
// MKS Robin 320x240 color display
//
//#define MKS_ROBIN_TFT ...does not compile
//#define PRINTER_EVENT_LEDS ...this causes problems when used in combination with previous line.
#define FSMC_GRAPHICAL_TFT**_

This does compile successfully. I think the TFT pins as definitions are not correct, as is needed for the Robin2. I'm trying to find where (in which file) to define the FSMC data pins. The data pins are not defined in the pins_MKS_ROBIN2.h file.

Ultimately I would prefer to use the official Marlin GitHub FW, as is. But it seems we are not there yet.

I also have problems with the M600 command option, needed for the Out of filament sensor. It requires one to define an LCD screen. The compiler does not seem to recognize having selected (defined) a TFT screen.

DaVincIan, 2019-08-31:
CJSoong

Did you get your Marlin 2.0 Robin2 FW to compile. I manged to get mine to compile, but there is something wrong. It seems as is the Fw upload does not complete. The screen shows progress up to 100 % but stops there. When I reboot, it shows booting... and that's all.

Fortunately I can re-flash the Robin2_V1.0.2 FM.

DaVincIan

3Dangel, 2019-08-31:
Ian,
It would be advisable to have the pinout from MKS’ GitHub and verify in
pins folder if the Robin2 pin assignment are correctly entered.
And compare it to the Robin’s.

This way, you may have better chance get the Robin2 working!
My cent

Cheers
Robert

cjsoong, 2019-09-01:
> CJSoong
>
> Did you get your Marlin 2.0 Robin2 FW to compile. I manged to get mine to compile, but there is something wrong. It seems as is the Fw upload does not complete. The screen shows progress up [...]
My upload is ok, but SD card & TFT is not working.
I have defined FSMC pin. But the compilation will be wrong.

#define LCD_RESET_PIN    PD13    //FSMC_RST
#define LCD_BACKLIGHT_PIN  PD12   //FSMC_LIGHT
#define FSMC_CS_PIN        PG12  // FSMC_NE4
#define FSMC_RS_PIN        PF12   // FSMC_A6

3Dangel, 2019-09-04:
there were a lot of changes on the GitHub where the Marling reference them to get libs from.

I just spent almos 8 hours yesterday to the late nigh and got the Marlin (not the 2.0xxbug) to compile and to get my touch and SD to work.
If there are interests here, I will them upload to the GitHub.

I'm not coder, just a persistant user and try hard to get the errors go away...:-0

cjsoong, 2019-09-05:
@3Dangel 
I need marlin 2.0 from MKS Robin2 V1.0.

3Dangel, 2019-09-05:
it won't work on yours!!!
mine is pre Robin2
mine was the very first batch MKS-Robin  V2.1 with TFT 1.1
it is NOT Robin2  mine has WIFI module soldered on the board so all pins
are different than yours

DaVincIan, 2019-09-06:
Hi Robert. Thank you for the time spent on this. 

Dit you upload the code that compile? 

I'm interested. Even if I have  a Robin2. 

Ian

DaVincIan, 2019-09-09:
CJSoong

Please share the folloing files:
platformio.ini
Configuration.h
Configuration_adv.h
pins_MKS_ROBIN2.h
and any other files you needed to change to get Robin2.bin to upload.

If I can get my Robin2 to upload, then we can work together on the SD & TFT.

Ian

cjsoong, 2019-09-10:
@DaVincIan 
https://drive.google.com/open?id=17-RYrc-iCv12Y3fG8ArDsCcz7IoZaJL3

DaVincIan, 2019-09-10:
Thank you CJSoon

I'll keep you posted

Ian

DaVincIan, 2019-09-10:
CJSoon

Again thank you for sharing...I'm starting to think there is something different on my Robin2 board/TFT controller. Your code compiled successfully, but do not upload. I tried uploading the Robin2.bin file included in your .rar file. Same problem. It shows green bar running up to 100 %, then it stops there. 

Super frustrating!!!

3Dangel, 2019-09-10:
Ian
That is my typical situation when using the fresh Marlin.
I think there’s something if the vast Marlin filed that alters the pin
assignments.
For instant, the spi.h in HAL- STM32xxxF1
It is assigned as PAx ...etc

So not sure what it is going in there.

Robin uses SPI2  to communicate.

But in the PlatformIO.ini
Under MKS-Robin env, the SPI’s lib is on the ignore list.

So far I cannot understand.

The only copy that it is “working”( if gone different layer of sub menus,
it gets jammed and only a reboot you can do at that point ) is from the
repository of JMZ52.

DaVincIan, 2019-09-11:
Robert

From CJSoon posts, I understood that he was able to config, compile and
successfully upload a Robin*2*.bin file that resulted in a working TFT
screen (no SD card or touch functionality though).

I looked/tried what he uploaded. That code will definitely will not
succeed. For starters, there is no TFT defined ...

From the Robin2 schematics, it seems as if the TFT controller is connected
to  SPI1 as the pin name is for instance (SPI1_MISO).
Further more, Robin2's
C:\Users\.platformio\packages\framework-arduinoststm32 is different. The
SPI library "SPI.cpp" does not have a SPIClass SPI(n) defined as under the
Maple framework. So I do not now where to set the code to compile for
SPI(2).

I'm not in a happy place...

Ian

3Dangel, 2019-09-11:
have you tried to select the right settings for the touch X calibration and
offsets?

/* MKS Robin TFT v2.0 */
//#define XPT2046_X_CALIBRATION    12013
//#define XPT2046_X_OFFSET           -32
//#define XPT2046_Y_CALIBRATION    -8711
//#define XPT2046_Y_OFFSET           256

/* MKS Robin TFT v1.1 */
#define XPT2046_X_CALIBRATION   -11792
#define XPT2046_X_OFFSET           342
#define XPT2046_Y_CALIBRATION     8947
#define XPT2046_Y_OFFSET           -19

these values are important for the touch to be "click" on at the right
position
this are located xpt2046.h

note this are not the UPFATED Marlin....
the updated Marlin uses other values and I have never got it to work

3Dangel, 2019-09-11:
and try to
#define SD_DETECT_PIN      -1

this should force the FW to ask you to manually ini the SD card

DaVincIan, 2019-09-11:
Robert

At this point in time I cannot get the FW upload to succeed. The
screen shoes a green band, runs 0 %...100 %, then stops (I suppose it
hangs). Rebooting shows" Booting..." and that is it.

My plan of action is to focus on getting the upload to succeed with
hopefully showing a working TFT. And from there (if not already
functioning) looking at solving SD card and Touch.

I just checked, there is no xpt2046.h for/under MKS-Robin2 project. It
seems to be part of STM32F1. Robin2 uses STM32F4 from what I can gather. It
looks more and more like the MKS-Robin2 MarlinFW branch is not a good
starting point to implement the Marlin 2.0 FW and that one should rather
start with Marlin 2.0bugfix  and figure out the configuration. From what I
have read on the MarlinFW Git posts on the subject, the  STM32F4
implementation is a work in progress and the best chance for success is
using VSCode.

Ian

3Dangel, 2019-09-11:
Ian
I would suggest you to start from makerbase’s GitHub.
It is not the most recent but it should have some sort of base done in
there.
Also, I would check what is the board version ( what is printed on your
board? Robin2 Vx.x)

cjsoong, 2019-09-11:
@DaVincIan 
I use the "pronterface" connection to determine the success of the upload, the time will be a long time, it should be the same as the initial failure when the SD card is initialized, but the connection will be successful. At this time, the sd card should be pulled out, so that each time the boot is turned on. Will not wait for a long time.

DaVincIan, 2019-09-14:
CJSoong

I managed to get the Robin2 to compile and upload, then use it with Repetier Host. I'm busy tuning the configuration for my printer. Once it is sorted I'll look into getting the TFT working.

Again, thanks for sharing your code.

3Dangel, 2019-09-14:
Congrats!!!
If it works with Repetier host then you might want to use it with Astro
print or Octoprint.
This way you could monitor the print from anywhere.

Cheers

DaVincIan, 2019-09-16:
Robert

I'm following your suggestion of using the MKS branch as a starting point.
My board is MKS Robin2 V1.0. I'm able to upload the FM. It ends with a
green bar across the TFT screen, but I have functionality using Repetier
Host.
I have similar success with CJSoong's branch. The code and configuration he
shared let me believe it is from Marlin 2.0 bugfix branch.

For now, I'm focusing on the MKS Robin2 Marlin 2 FW branch.

There seems to be quite a number of issues to resolve. I believe 1) to 3),
is mostly pins definition related:
1) The TFT display (looks like a FSMC based controller and I'm
investigating the generic_STM32 F407 pins)
2) Touch screen
3) SD card
4) BLTouch. At the moment it is "dead". It initializes when the power is
applied, but nothing after that.
5 Dual Z stepper motors (stepper Z and stepper E1. I have both working.
They work fine in the Z+ direction, but when I move Z- stepper Z turns in
the same direction as when I do Z+ move. But stepper Z (E1), turns correct
direction.

Ian

innc1, 2019-09-17:
Ian I'm looking forward to see a working TFT display at robin2. This has been busy me for nearly two weeks. Please inform us asap. Thanks for replys