makerbase-mks/MKS-Robin #124: TFT35 MKS-Robin Nano white Display

ph1lloops, 2019-08-28:
I flashed Marlin 2.0 and on the Robin Nano 1.1 and also the MKS Robin TFT35 V1.0 Firmware for the Display. After Booting the display keeps white and nothing happens. Connecting to the MKS Robin Nano with pronterface works. Do I have to set something else in Marlin?

zillarob, 2019-08-28:
You dont need fw for the screen, its build into Marlin.
You do need to enable it and the touch buttons in Marlin.
Look for the FMSC stuff down toward the bottom of config.h.

ph1lloops, 2019-08-29:
Hey,
Thanks for you help. I was searching in Marlin in configuration.h but I can not find anything with FMSC. Maybe you can send me a screenshot of the part you mean.

Thanks!
<Philipp

DaVincIan, 2019-08-29:
Start by in the config.h file search for TFT.

ph1lloops, 2019-08-29:
[image]

zillarob, 2019-08-29:
I forgot they have an odd version here.
It used to say that, but the newer version is slightly different now.
Id grab a fresh version from the main branch. Only use the ones here as a template for some of the stuff like serial settings etc.

ph1lloops, 2019-08-29:
Using the fresh version with the nano board selected and the TFT aktivated = display stays completely off

zillarob, 2019-08-29:
There is no tft option in the main marlin branch.

[image]

ph1lloops, 2019-08-29:
That’s the line changed like in your screenshot. What is also missing in my firmware?

LG Philipp

[nan]<https://user-images.githubusercontent.com/42916376/63910010-0636c780-c9f3-11e9-8930-6ef6eee01ecb.JPG>

zillarob, 2019-08-29:
Hard to say.
Ive been playing with it on my Tevo Nereus and the toughest thing I remember was getting the buttons to work.
It does seem a lil buggy though. I cant say if its the fw, the board, or me doing something goofy in the fw though.
It does seem like if I dl and setup a new it will be fine, but as time goes by and I turn stuff on or off it just gets worse and worse.

zillarob, 2019-08-29:
You could try adding this if not there
[image]

ph1lloops, 2019-08-29:
Wow thanks for your help. I’ll try that when I am back home. Just add that to the end of configuration.h right? Just to make sure: I changed the Board and the Ports + uncommented the TFT line like in your screenshot. Anything else I have to take care of?

LG Philipp

[nanpin]<https://user-images.githubusercontent.com/42916376/63910408-945f7d80-c9f4-11e9-8fef-5b0abb696a4d.JPG>

zillarob, 2019-08-29:
You need to see what is already in there, and then add the stuff that isnt. 
Dont just paste that at the bottom.

I dont know what you will have to change for your printer, but that is what I use and the screen and touch buttons work.

ph1lloops, 2019-08-29:
Yes sure I’ll change all the stuff for my printer but first I wanted to make the display to work.

LG Philipp

alfrankgit, 2019-08-29:
You need to uncomment this in PINS_MKS_ROBN_NANO.h so that that 3.5“ display gets properly initialized:

//#define MKS_ROBIN_TFT35     //Uncomment when using the robin tft35 display

ph1lloops, 2019-08-29:
Is that in the configuration.h that line?

LG Philipp

alfrankgit, 2019-08-29:
It is in the file PINS_MKS_ROBN_NANO.h as I wrote above.

ph1lloops, 2019-08-29:
Yay it works now. The only thing: the TFT shows the Marlin layout. What do I need to change to get the MKS TFT design?
Big big thanks!
[image]

zillarob, 2019-08-29:
I dont think you can.
You either get an updated version of marlin with the marlin menus,
or an old version of marlin with mks menus and who knows wtf else going on because mks never released the source. (if you look at it with a hex editor it was way old, like 1.0)

That version you are using from here is getting pretty old now also (2.0 at least), and the touch buttons arent nearly as responsive as the current main branch iirc.

ph1lloops, 2019-08-29:
Ah ok. I was looking for the 
define MKS_ROBIN_TFT35 //Uncomment when using the robin tft35 display
in the newest 2.0.x Marlin firmware, but I can not see it there. How can I activate the TFT35 in that newer Marlin version?
Thanks again for your help and work!

alfrankgit, 2019-08-29:
No, the MKS style menu is only in the precompiled Release 1.0.2, there’s no source code for that.
I think the Marlin 2.0 from MKS is just for alibi.

ph1lloops, 2019-08-29:
Ok I flashed the newest marlin and the display runs after uncommenting 
//#define LCD_RESET_PIN      PF6

The touch button are missing and the resolutions seems to be not correct. These two things and i am happy. Do have an idea how to fix those last two things?

zillarob, 2019-08-29:
Never seen that res thing.
Enable the touch buttons and maybe the other stuff I mentioned above.

zillarob, 2019-08-29:
Chip on nano doesnt have PF6. I think that was supposed to be PC6 for nano.

Technology89, 2019-09-01:
Tried the same as Phillip today and got exactly the same result. Display just shown in odd resolution and no touch support.

3Dangel, 2019-09-01:
I would compare the pins assignment and how it is configured in mks-Robin
and verify the pinout from the schematic.
Mks-Robin touch is working.

Check the issue case in mks-robin ( use key word on Google “ mks Robin
Marlin”)

A gentlemen posted a small FW to test what LCD controller the board is
using.

It seemed to me that for the moment, there is only couple lcd controller
that are supported.

Hope this helps.

zillarob, 2019-09-01:
I dont know if this is what it is using, but I see quite a few changes in the tft upscaler thing in lcd dogm between the older one I am using and the current version.
If I get bored this eve, I will do up a freshy and see if I see the same.

Technology89, 2019-09-02:
Got the touchbuttons to show up but they don't work. Didn't define TOUCH_BUTTONS in configuration.h

https://i.imgur.com/yTqXn7Lr.jpg

3Dangel, 2019-09-02:
Yours is the latest version that they just forked to.
Mine worked with the green original settings.

Thanks for advise as I am about to upload the newest updated FW as yours.
Color is good!!!!
Cheers

3Dangel, 2019-09-02:
Hello Technology89,

 Would it be possible to get zip file from you?
I’m having issue with my just downloaded version of Marlin.
The old one that I had for s month or so still compiles correctly ( without
your color buttons)

Thanks in advance.
Bob

Technology89, 2019-09-02:
Gonna upload tomorrow. But remember that the buttons not working. They are just shown but not touchable.

3Dangel, 2019-09-02:
Thanks man…
Will compare what I have
Compiling now with my setttings..

Thanks v much!!!
Robert

From: Technology89
Sent: Monday, September 2, 2019 1:58 PM
To: makerbase-mks/MKS-Robin
Cc: 3Dangel; Comment
Subject: Re: [makerbase-mks/MKS-Robin] TFT35 MKS-Robin Nano white Display(#124)

Gonna upload tomorrow. But remember that the buttons not working. They are just shown but not touchable.

Technology89, 2019-09-03:
Here you go. Maybe you need to tune some settings as i am using LV8729 for Z and E instead of the a4988

https://github.com/Technology89/Marlin_2.0_Sapphire_Pro

3Dangel, 2019-09-03:
Thanks!!
I got it last night to work my touch.
Not from the latest Marlin.
No color, just green standard.
And SD works as well.
It was a heck of work an try on error!!

Technology89, 2019-09-03:
But still marlin 2.0? SD eeprom also working? Mind sharing your version?

3Dangel, 2019-09-03:
mine is the Robin
not sure about eeprom
only got it to load and checked my menus.

I will post like your did in github tonight

but short thing for your to try is the Touch_buttom calibration

those values are most likely off

these are my values

 #define XPT2046_X_CALIBRATION   -11876
  #define XPT2046_Y_CALIBRATION  8755
  #define XPT2046_X_OFFSET       346
  #define XPT2046_Y_OFFSET        -20
kind of flipped from what it is on Marlin.

I did use a calibration tool from a gentleman who worked extensively for
Robin...

and my version is older ..
the new one gives me black screen, so it is something not right.....
need to check yours to compare since the NANO is very similar to my
Robin.....

cheers
keep in touch

(working with another gentleman from Sought Africa as well)
me from Calgary, AB Canada

Robert

Technology89, 2019-09-03:
Gonna try these values tomorrow but you have a '#' in front of these values so they are not assigned

3Dangel, 2019-09-03:
It is all in one line
 #define XPT2046_X_CALIBRATION   -11876

3Dangel, 2019-09-03:
I see what you are talking about

You need the “#”

It is the “//“ that uncomment the line

Technology89, 2019-09-03:
Of course... Seems like i had a brainfart 😄

zillarob, 2019-09-04:
Finally got to try a fresh version out, looks ok.
Really like the new cancel button. Saves a bunch of time in the menus.

I think the toughest part on the new versions is the pins file. This is one I have been using for a while and seems to be ok, but have been being lazy and just rolling stuff into it lately. Not sure how much is actually needed in the new stuff or if there is some redundancy, but I should go back and start fresh one of these days and find out how much is needed.
This is for Main branch Marlin after you setup config and config.h for whatever you have.
Might need to change a few pins also, I run BLTouch so some stuff there is swapped around.
[pins_MKS_ROBIN_NANO.zip](https://github.com/makerbase-mks/MKS-Robin/files/3572664/pins_MKS_ROBIN_NANO.zip)

zillarob, 2019-09-04:
The X on the left is nice. 
No more having to scroll up to the top to go to the previous menu, just hit the X.

[image]

3Dangel, 2019-09-04:
Cool staff!!!
I was not able to make my Robin v2.1 to work yet under Marlin2 xxxbug yet.
I got it to compile and get the .bin but the screen stays black.

Thinking that it has to be pins issues.
Thanks for sharing yours.
Mine “works” ( not hooked up yet with motors and etc) under an older
brancknofnMarlin

zillarob, 2019-09-04:
Im on a Nano so no idea how it will work on the other boards.

Technology89, 2019-09-04:
@zillarob can you upload the whole marlin? Your screen is fullscreen and not only the left corner like with mine version. I don't think this is only a pins issue on mine.

zillarob, 2019-09-04:
Doesnt look like, too big.

zillarob, 2019-09-04:
Here is the compiled one if you want to test, but dont try to print with it.
[Robin_nano.zip](https://github.com/makerbase-mks/MKS-Robin/files/3572767/Robin_nano.zip)

3Dangel, 2019-09-04:
Zillarob,

Not sure if it is too much too ask:
Would it be possible to have the marlin config (x2) and thenplatformIO.ini
from you?

I know mine is not nano.
But they both share much in common. So I could compare the two and
hopefully I could find the way out  of it.

Thanks in advance

zillarob, 2019-09-04:
Only thing in pio.ini is setting the env to robin nano.
[configs.zip](https://github.com/makerbase-mks/MKS-Robin/files/3572795/configs.zip)

3Dangel, 2019-09-04:
thanks a lot!
it all due to the lib dependency that I saw I am having...
the TMC was givcing me tons of issues and I had to delete some of the
source and replace it with the Bigtree's TMC lib to get the compile going.

Technology89, 2019-09-04:
You can upload here on github 👌

Technology89, 2019-09-04:
Thank you so much @zillarob ! The touchbuttons are working but my display is still only small in the top left corner while the touch buttons are sensitive where they used to be...

Can you upload your u8g_dev_tft_320x240_upscale_from_128x64.cpp ? Seems like this one is also different...

zillarob, 2019-09-04:
That file is untouched from the main marlin branch.

Technology89, 2019-09-05:
Strange... The touchscreen is scaled correct only the lcd is scaled wrong. It just shown as 128*64 pixels. Also the sd card eeprom won't save. Gonna have a second look at your configuration.h later

zillarob, 2019-09-05:
I have been messing with the sd stuff. Not positive where I left off, but there was an eeprom.dat on there when I flashed it.

zillarob, 2019-09-05:
I think I remember seeing that they used a few different chips on the tft. Maybe yours is different?
I never had to mess with it so not if needed to tell it which one or something.

zillarob, 2019-09-05:
Ya, have eeprom.dat back on there so gonna be you config or config.adv.
I delete it before I flash.

3Dangel, 2019-09-05:
cool man!
if there was an issue and you fixed it, worth to post it to the pull on
marlin so it gets merged into it!!!!
What was the setting you made to have it to work?
cheers

zillarob, 2019-09-05:
The eeprom thing is just a standard marlin setup thing.
When you compare my config and config.adv you will prob find it.

3Dangel, 2019-09-05:
ok got it!!
been working to get the screen .....
not sure what they did  but mine is either on booting or white....
well time to sleep.... thanks again!!!!

Technology89, 2019-09-05:
@3Dangel when your screen is white uncomment '#define LCD_RESET_PIN      PF6'. Then it should work.

@zillarob gonna try your .bin file today and will test if my lcd scales up.

[55 more replies]