makerbase-mks/MKS-Robin #125: MKS-Robin-Nano Marlin2 compile error

sacy80, 2019-09-01:
i Try to compile marlin2 for robin nano but there error. using the fw from makerbase-mks/MKS-Robin/MKS-Robin Nano/frimware/Marlin2.0-MKS-Robin_nano

i have also try marlin2.0 bugfix from marlin also have compiler error .

ERROR INFO:
Arduino-L6470 @ 0.7.0 has been successfully installed!
LibraryManager: Installing c1921b4
Using cache: C:\Users\sacy.SACY\.platformio\.cache\8c\f244bfa38b7bfff5e9042f5d9f9f5c8c
Unpacking...

**IOError: [Errno 2] No such file or directory: 'C:\\Users\\sacy.SACY\\Downloads\\MKS-Robin-master\\MKS Robin Nano\\firmware\\Marlin2.0-MKS-Robin_nano\\.piolibdeps\\mks_robin_nano\\_tmp_installing-e8_nsw-package\\TMC26XStepper-c1921b4856a5ef4b659b661b8cb5dd56f3075e57\\documentation\\htm
l\\_t_m_c26_x_stepper_8cpp_source.html':
  File "C:\Users\sacy.SACY\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 126:**
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\sacy.SACY\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\sacy.SACY\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
    exec _file_ in call_stack[-1].globals
  File "C:\Users\sacy.SACY\.platformio\platforms\ststm32\builder\main.py", line 91:
    target_elf = env.BuildProgram()

  File "C:\Users\sacy.SACY\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\sacy.SACY\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 122:
    _build_project_deps(env)
  File "C:\Users\sacy.SACY\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 47:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\sacy.SACY\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\sacy.SACY\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1043:
    project.install_dependencies()
  File "C:\Users\sacy.SACY\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 878:
    lm.install(uri)
  File "c:\users\sacy.sacy\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 359:
    force=force)
  File "c:\users\sacy.sacy\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 701:
    track=True)
  File "c:\users\sacy.sacy\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 483:
    self.unpack(dlpath, tmp_dir)
  File "c:\users\sacy.sacy\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 216:
    raise e
 
 [FAILED] Took 12.77 seconds
Environment           Status    Duration
--------------------  --------  ------------
megaatmega2560        IGNORED
megaatmega1280        IGNORED
at90usb1286_cdc       IGNORED
at90usb1286_dfu       IGNORED
DUE                   IGNORED
DUE_USB               IGNORED
DUE_debug             IGNORED
LPC1768               IGNORED
LPC1769               IGNORED
melzi                 IGNORED
melzi_optiboot        IGNORED
rambo                 IGNORED
sanguino_atmega644p   IGNORED
sanguino_atmega1284p  IGNORED
STM32F1               IGNORED
STM32F4               IGNORED
ARMED                 IGNORED
mks_robin             IGNORED
mks_robin_nano        FAILED    00:00:12.775
teensy35              IGNORED
malyanm200            IGNORED
esp32                 IGNORED
fysetc_f6_13          IGNORED
 1 failed, 0 succeeded in 00:00:12.775

cjsoong, 2019-09-02:
@sacy80 
The directory path is too long.

sacy80, 2019-09-02:
nice thanks a lot. try and it can compile

closed 2019-09-02 by sacy80