makerbase-mks/MKS-Robin #127: MKS-Robin2 with TMC2208 compile error

aknalee, 2019-09-06:
Compiling with VS code and PlatformIO

I set the 
-----------(platformio.ini)
[env:mks_robin2]
platform = ststm32@5.4.3 

-----------(configuration.h)
#define Y_DRIVER_TYPE  TMC2208

then I got the error below.. please help me!!

Marlin\src\HAL\HAL_STM32\../../inc/SanityCheck.h:1958:4: error: #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."
   #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."
    ^~~~~
In file included from Marlin\src\HAL\HAL_STM32\../../inc/MarlinConfig.h:37:0,
                 from Marlin\src\HAL\HAL_STM32\HAL_Servo_STM32.cpp:25:
Marlin\src\HAL\HAL_STM32\../../inc/SanityCheck.h:1958:4: error: #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."
   #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."
    ^~~~~
In file included from Marlin\src\HAL\HAL_STM32\../../inc/MarlinConfig.h:37:0,
                 from Marlin\src\HAL\HAL_STM32\HAL_spi_STM32.cpp:26:
Marlin\src\HAL\HAL_STM32\../../inc/SanityCheck.h:1958:4: error: #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."
   #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."
    ^~~~~
In file included from Marlin\src\HAL\HAL_STM32\../../inc/MarlinConfig.h:37:0,
                 from Marlin\src\HAL\HAL_STM32\fastio_STM32.cpp:25:
Marlin\src\HAL\HAL_STM32\../../inc/SanityCheck.h:1958:4: error: #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."
   #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."
    ^~~~~
In file included from Marlin\src\HAL\HAL_STM32\../../inc/MarlinConfig.h:37:0,
                 from Marlin\src\HAL\HAL_STM32\HAL_timers_STM32.h:25,
                 from Marlin\src\HAL\HAL_STM32\HAL.h:99,
                 from Marlin\src\HAL\HAL_STM32\HAL_timers_STM32.cpp:25:
Marlin\src\HAL\HAL_STM32\../../inc/SanityCheck.h:1958:4: error: #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."
   #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."
    ^~~~~
In file included from Marlin\src\HAL\HAL_STM32\../../inc/MarlinConfig.h:37:0,
                 from Marlin\src\HAL\HAL_STM32\persistent_store_impl.cpp:25:
Marlin\src\HAL\HAL_STM32\../../inc/SanityCheck.h:1958:4: error: #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."
   #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."
    ^~~~~
In file included from Marlin\src\HAL\HAL_STM32\../../inc/MarlinConfig.h:37:0,
                 from Marlin\src\HAL\HAL_STM32\watchdog_STM32.cpp:25:
Marlin\src\HAL\HAL_STM32\../../inc/SanityCheck.h:1958:4: error: #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."
   #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."
    ^~~~~
In file included from Marlin\src\HAL\shared\../../inc/MarlinConfig.h:37:0,
                 from Marlin\src\HAL\shared\HAL_spi_L6470.cpp:28:
Marlin\src\HAL\shared\../../inc/SanityCheck.h:1958:4: error: #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."
   #error "TMC2208 or TMC2209 on Y requires Y_HARDWARE_SERIAL or Y_SERIAL_(RX|TX)_PIN."

makerbase-mks, 2019-10-11:
Since you have defined Y-AXIS to use TMC2208, you need to add the following pin definitions in the pins_MKS_ROBIN.h file:
 #if HAS_TMC220x
  /**
   * TMC2208/TMC2209 stepper drivers
   *
   * Hardware serial communication ports.
   * If undefined software serial is used according to the pins below
   */
  //#define X_HARDWARE_SERIAL Serial
  //#define X2_HARDWARE_SERIAL Serial1
  //#define Y_HARDWARE_SERIAL Serial1
  //#define Y2_HARDWARE_SERIAL Serial1
  //#define Z_HARDWARE_SERIAL Serial1
  //#define Z2_HARDWARE_SERIAL Serial1
  //#define E0_HARDWARE_SERIAL Serial1
  //#define E1_HARDWARE_SERIAL Serial1
  //#define E2_HARDWARE_SERIAL Serial1
  //#define E3_HARDWARE_SERIAL Serial1
  //#define E4_HARDWARE_SERIAL Serial1

  //
  // Software serial
  //
  #define X_SERIAL_TX_PIN ***
  #define X_SERIAL_RX_PIN ***

  #define Y_SERIAL_TX_PIN ***
  #define Y_SERIAL_RX_PIN ***

  #define Z_SERIAL_TX_PIN ***
  #define Z_SERIAL_RX_PIN ***

  #define E0_SERIAL_TX_PIN ***
  #define E0_SERIAL_RX_PIN ***

  #define E1_SERIAL_TX_PIN ***
  #define E1_SERIAL_RX_PIN ***
#endif

*** Represents your pin, such as PF1... You can refer to the MKS Robin schematic and select the appropriate extension pin.
MKS Robin Hardware: https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin/firmware

makerbase-mks, 2019-10-11:
Since you have defined Y-AXIS to use TMC2208, you need to add the following pin definitions in the pins_MKS_ROBIN.h file:
 #if HAS_TMC220x
  /**
   * TMC2208/TMC2209 stepper drivers
   *
   * Hardware serial communication ports.
   * If undefined software serial is used according to the pins below
   */
  //#define X_HARDWARE_SERIAL Serial
  //#define X2_HARDWARE_SERIAL Serial1
  //#define Y_HARDWARE_SERIAL Serial1
  //#define Y2_HARDWARE_SERIAL Serial1
  //#define Z_HARDWARE_SERIAL Serial1
  //#define Z2_HARDWARE_SERIAL Serial1
  //#define E0_HARDWARE_SERIAL Serial1
  //#define E1_HARDWARE_SERIAL Serial1
  //#define E2_HARDWARE_SERIAL Serial1
  //#define E3_HARDWARE_SERIAL Serial1
  //#define E4_HARDWARE_SERIAL Serial1

  //
  // Software serial
  //
  #define X_SERIAL_TX_PIN ***
  #define X_SERIAL_RX_PIN ***

  #define Y_SERIAL_TX_PIN ***
  #define Y_SERIAL_RX_PIN ***

  #define Z_SERIAL_TX_PIN ***
  #define Z_SERIAL_RX_PIN ***

  #define E0_SERIAL_TX_PIN ***
  #define E0_SERIAL_RX_PIN ***

  #define E1_SERIAL_TX_PIN ***
  #define E1_SERIAL_RX_PIN ***
#endif

*** Represents your pin, such as PF1... You can refer to the MKS Robin schematic and select the appropriate extension pin.
MKS Robin Hardware: https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin/Hardware

pavlov-s, 2020-06-18:
@makerbase-mks 
How about you go into your own repository and check how your recommendations apply your own design ?
Neither Robin nor Robin2 support TMC UART.
The UART_PDN pin of the driver connects to the socket' SLP line which goes nowhere.
End of story.