makerbase-mks/MKS-Robin #128: complied firmware is not updating via SD card

github051, 2019-09-07:
I have complied the marlin2 firmware successfully and i copied the firmware file to sd card and plug into MKS Robin board. But it is not updating. any idea why ?

cjsoong, 2019-09-07:
@github051 
Change the file name to firmware.bin.

3Dangel, 2019-09-07:
Really?
I have never try that.
I have always copied the Robin.bin
Make sure your power cycle with the SD installed.
Sometime it won’t detect SD insertion due to bad FW ( eg not pun assign for
the SD_detect)

github051, 2019-09-07:
i tried my self. fw is updated but lcd is not displaying anything

github051, 2019-09-07:
how to change jerk settings with M code ?

i tried with M205 X10 and enter after that again M500 to store vales in EPROM
but this is also not working

can anybody tried in MKS Robin ?

github051, 2019-09-09:
> i tried my self. fw is updated but lcd is not displaying anything

Finally its working

closed 2019-09-09 by github051