makerbase-mks/MKS-Robin #132: How to change Jerk settings in Mks robin ?

github051, 2019-09-09:
Hi Makerbase,
How to change Jerk settings in Mks robin ?
Default jerk values are high. How to change it ? Lot people are facing this issue. Why you are not doing anything for this?

Simple code change also you are not updating. 
Worst board and work service. Very bad experience with Mks robin. Also with MKS team. they never reply or they never respond. what the hell is this.

If you are not able to support then give the source code.

washeree, 2019-09-10:
Do NOT use their software as you see there is no support nothing. Also their software closed which 3d printer community doesnt like. 

Just install 32bit Marlin and Enjoy. Dont waste your time.

Cheers

github051, 2019-09-10:
> Do NOT use their software as you see there is no support nothing. Also their software closed which 3d printer community doesnt like.
> 
> Just install 32bit Marlin and Enjoy. Dont waste your time.
> 
> Cheers

Thanks for that.
I flashed with marlin 2.0 bugfix. Again here TFT screen buttons are not user friendly.
Basically i'm a mechanical engineer. Understanding all those this is very difficult.
Because of that reason i'm seeking their help.

makerbase-mks, 2019-10-09:
If you are using Robin firmware, this parameter cannot change. We will consider the new firmware can change it in the future.

github051, 2019-10-09:
> If you are using Robin firmware, this parameter cannot change. We will consider the new firmware can change it in the future.

When can we expect the New version of Robin firmware ?

makerbase-mks, 2019-10-11:
I have learned about this part of the demand and we will speed up the process.

github051, 2019-10-11:
I think from last 6 moths there is no version upgrade. Please concentrate on MKS Robin and improve the firmware.