makerbase-mks/MKS-Robin #133: Mks Robin2 Circles printing issue.

github051, 2019-09-11:
Mks Robin2 problems:
1. Printing via USB is not working properly. It is stopping same location even if you restart the printer.
2. While printing circles lot of click noise is observed from motors.

I am using latest firmware of robin2.

makerbase-mks, 2019-10-11:
1. I want to confirm whether you can use the host computer to control the motherboard and realize the motor action...
2, the motor noise is large, this has a lot to do with the drive.
3, for the second question, I recommend you use MKS TMC2208 or MKS TMC2209, this driver, the mute effect is very good.
MKS TMC2208：https://www.aliexpress.com/item/32888980385.html?spm=2114.12010612.8148356.1.5fba18b7vGEtua
MKS TMC2209：https://www.aliexpress.com/item/33043140087.html?spm=2114.12010612.8148356.47.5fba18b7Ohqp2m

github051, 2019-10-11:
Reg Motor Noise issue: I am printing via SD card and also i am using Mks TMC2208 stepper drivers with 1/32 micro steppering.
Still the noise is observed only when i am printing Circles.
Normal lines are printing ok.

Same settings and same G-Code and Same TMC2208 drives with MKS Robin is printing with out any Noise.