makerbase-mks/MKS-Robin #136: Firmware for Tevo Nereus

jkoenig72, 2019-09-21:
Can somebody point me to a working firmware for the Tevo Nereus?

3Dangel, 2019-09-21:
If you are new to 3D printing world, please be prepare on your own a lot of time. 
There’s no FW for particular printer. 
You make your own from the stock. 
You could try Marlin. 
Just spend time to try in error. 

Good luck

zillarob, 2019-11-30:
In case you havent found it in the group already.
https://github.com/thisiskeithb/Marlin/tree/test_tevo_nereus/Marlin

Dan-Knox, 2020-01-18:
@zillarob: it appears as though that branch has been deleted. Can you point me to another copy?

jlawless6925, 2020-01-31:
I could use a copy as well.... It's really hard to find a working English copy and my SD card that came with the printer somehow got corrupted

fjd215, 2020-02-29:
Ditto, very difficult to deal with Tevo support and get your hands on the original firmware.