makerbase-mks/MKS-Robin #137: How to connect auto off?

shivahegde, 2019-09-23:
I bought Robin mini board, but where and how to connect auto off button to board?
[image]

cjsoong, 2019-09-24:
@shivahegde 
Auto power off requires a unlock switch and a relay board
Or use the ATX power supply.
[image]

makerbase-mks, 2019-10-11:
MKS Robin Mini connect auto off button by SW，You need to use the NO end of the non-self-locking switch to connect to the SW of the MKS Robin Mini.

shivahegde, 2019-12-05:
I connected as you suggested, but it wont auto off after printing

github051, 2020-05-18:
Issue is still there ?