makerbase-mks/MKS-Robin #138: MKS Robin touch screen not working with Marlin 2.0

Slewentogzz, 2019-10-01:
I followed the instructions on how to install Marlin 2.0, it went smoothly and everything works. The display also works but I can't control it with touch, it's like a normal display without touch functionality. Anyone know how to solve this?

3Dangel, 2019-10-01:
Great for you !!

Just you need to alter the calibration X/Y and it’s offset.
Try that way.

Slewentogzz, 2019-10-01:
I don't understand how that will make the touch functionality work? I want to be able to use the screen, otherwise I can't print with a SD card

Slewentogzz, 2019-10-01:
Where do I find the setting to change this?

3Dangel, 2019-10-01:
In the configuration.h

You should have enabled the touch already.
Depending what version you have and from where you downloaded it ( Marlin)
I recall mine was just under the display section around the SD card
settings or the pins_Mks Robin.h

3Dangel, 2019-10-01:
Refer to this post and find the little FW specifically that runs for
calibration. And you should see what is your values.

Use it at your own risk.
I used it and it worked for me.

https://github.com/MarlinFirmware/Marlin/issues/9771

Slewentogzz, 2019-10-01:
Allright, thanks. In the firmware provided by MKS the x-y-settings didn't exist. I've downloaded the default marlin now and the settings you are talking about is in the config file. However it doesn't compile, it says "CS_PIN not declared in this scope" (in file Marlin\src\feature\touch\xpt2046.cpp).

3Dangel, 2019-10-01:
Yeah, Marlin branch does work.
The one that worked for me was in the Makerbase’s.
I now don’t remember where the x-y values needs to be.
I need to get back home and search mine.
It will be later today when home from work.

3Dangel, 2019-10-02:
check the MKS-Robin <https://github.com/makerbase-mks/MKS-Robin>/MKS Robin
<https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin>/
firmware
<https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin/firmware>
/Marlin2.0-MKS-Robin
<https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin/firmware/Marlin2.0-MKS-Robin>
/Marlin
<https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin/firmware/Marlin2.0-MKS-Robin/Marlin>
/src
<https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin/firmware/Marlin2.0-MKS-Robin/Marlin/src>
/HAL
<https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin/firmware/Marlin2.0-MKS-Robin/Marlin/src/HAL>
/HAL_STM32F1
<https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin/firmware/Marlin2.0-MKS-Robin/Marlin/src/HAL/HAL_STM32F1>
/xpt2046.h

in there you need to select what type of LCD you have and then play with
the "-" (negative sign) I believe it has to be negative x offset and
negative Y calibration

try and see

*good luck*

Slewentogzz, 2019-10-02:
THANK YOU! Now the touch works. It was setup for a different display version, I just had to comment some things out in xpt2046.h. However the SD card doesn't show up, but that's a different issue.

closed 2019-10-02 by Slewentogzz

3Dangel, 2019-10-03:
For the SD card, follow the same link I sent you and towRd the bottom of
the post , change your SD_ pins  and your should see that you can manually
activate the SD card.