makerbase-mks/MKS-Robin #139: MKS Robin Lite extruder motor issue

freez10, 2019-10-08:
Hello guys.
I'm trying to build a firmware out of this source for my MKS Robin Lite and I have one issue. Everything seems working fine except for the extruder motor. It doesn't push the filament. It spins without filament but looks like has no torque since it can be stopped with a finger.
With Lite_0.0.5.rar and Robin Lite_0.0.4 Release file.zip everything works fine, so there is no problem from point of the hardware.

I tried to play with DEFAULT_PWM_MOTOR_CURRENT and it seems it has no effect on E. Cause if I eg set it to 10 for Z I can easily stop it, for example. But changes of third part, that should control E current have no effect.

Could you give me a clue where I should start looking?

makerbase-mks, 2019-10-09:
You can try the following settings (in the pins_MKS_ROBIN_LITE.h file)
#define MOTOR_CURRENT_PWM_RANGE (65535/10/3.3)
#define DEFAULT_PWM_MOTOR_CURRENT { 1000, 1000, 1000 }

freez10, 2019-10-09:
I tried, thanks. Any changes of anything connected to E0 current have no effect. But sorry guys, it was by mistake. With sources you provide here E0 works. Screen is mirrored but I know how to fix this, so it's ok. E0 doesn't work in the latest Marlin fw bugfix-2. 
Btw is it you who supports MKS stuff in https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x ?
if so, maybe you could help me solving this issue there? If not I'm closing the issue. 
Thanks

makerbase-mks, 2019-10-31:
This is may be a bug for Marlin2.0, we will continue to pay attention.

freez10, 2019-10-31:
I stopped struggling with it, this board does not worth that time, but here is what I found:
If MKS_MINI_12864 is enabled the screen works but extruder motor doesn't. And vice versa. Have no idea why. Pins configuration seems to be correct but it doesn't work.

tuncof, 2021-06-25:
How did you manage to compile Marlin 2.0 for MKS Robin Lite? Do you have the binaries? Regards.

freez10, 2021-06-26:
> How did you manage to compile Marlin 2.0 for MKS Robin Lite? Do you have the binaries? Regards.

I didn't