makerbase-mks/MKS-Robin #140: MKS Mini Z-Probe configuration

3DBeerGoggles, 2019-10-20:
So from what I've read the only supported Z-probe for auto bed level is one that uses the Z-Stop circuit - so assuming for a moment that I have a sensor set up to close the switch as designed, etc. - how do I configure this in firmware?

The configuration file has no where to enter a Z-probe offset, enable "safe" Z zeroing (ensuring the probe is over the bed when zeroing).

When I enabled autoleveling using Z-probe as a test (assuming my printer would just hit the end stop and "think" it had probed the bed), it tested in a pattern but quickly ran outside the bed limit.

So I suppose this could be summed up as "Is there any guide or comprehensive information on setting up Z-probe autolevel on the MKS Robin Mini?"

Guilbaut, 2019-10-20:
+1  same problem with Robin nano
the lack of a solution for interfacing z sensors and a serious handicap for the use of this controller

makerbase-mks, 2019-10-22:
MKS Robin Mini or MKS Robin Nano set leveing in config file:
1、Leveling method set
>cfg_leveling_mode			1    #（1:AUTO    0:Manual）
2、If use Manual leveling mode setting parameters
>cfg_point_number	5	
>cfg_point1:50,50						
>cfg_point2:180,50
>cfg_point3:180,180						
>cfg_point4:50,180
>cfg_point5:100,100
3、If use AUTO leveling mode setting parameters
>cfg_auto_leveling_cmd:G28;G29;
4、Set the interface used (for example, using Z+)
>USE_XMIN_PLUG	0			# 1:use; 0:unuse
>USE_YMIN_PLUG	0			# 1:use; 0:unuse
>USE_ZMIN_PLUG	0			# 1:use; 0:unuse
>USE_XMAX_PLUG	0			# 1:use; 0:unuse
>USE_YMAX_PLUG	0			# 1:use; 0:unuse
>USE_ZMAX_PLUG	1			# 1:use; 0:unuse
5、finish set parameters

3DBeerGoggles, 2019-10-22:
My config file doesn't have any "USE_XMIN_PLUG" etc. entries. Also, how does one set the Z-probe offset, or to make sure the probe moves over the bed before zeroing the Z-axis?

Could you perhaps post an example config with Z-probing set up on the Robin mini?

3DBeerGoggles, 2019-10-27:
So.... any update on this?

3DBeerGoggles, 2019-12-05:
Well, call me underwhelmed at the level of documentation & support Makerbase provides. Why make a closed-source firmware if you're not going to put effort into communicating how it's used.