makerbase-mks/MKS-Robin #141: MKS robin marlin compilation error

github051, 2019-10-23:
Can anybody please help me out on this issue ?
I have attached my config files
[image]

[Configuration.zip](https://github.com/makerbase-mks/MKS-Robin/files/3761510/Configuration.zip)

3Dangel, 2019-10-23:
Try to shorten your folders.
Put your Marlin folder @ C:!root and try

github051, 2019-10-23:
Thanks for reply.
After placing the marlin folder in C: root directory also same issue .

TimPut, 2019-10-23:
This is an issue with python3 being stricter about distinguishing byte strings and char strings. Here's the diff from my local checkout which fixes it for me:
```
diff --git a/buildroot/share/PlatformIO/scripts/mks_robin_nano.py b/buildroot/share/PlatformIO/scripts/mks_robin_nano.py
index eb8bca9..8b36d23 100644
--- a/buildroot/share/PlatformIO/scripts/mks_robin_nano.py
+++ b/buildroot/share/PlatformIO/scripts/mks_robin_nano.py
@@ -21,7 +21,8 @@ def encrypt(source, target, env):
         while position < length:
             byte = firmware.read(1)
             if position >= 320 and position < 31040:
-                byte = chr(ord(byte) ^ key[position & 31])
+                byte = ord(byte) ^ key[position & 31]
+                byte = byte.to_bytes(1, byteorder='big')
             robin.write(byte)
             position += 1
     finally:
```

github051, 2019-10-24:
HI TimPut,
Thanks for reply. After modification as per your code it's working. Once again big thanks.:)

github051, 2019-10-25:
i am able to compile the robin firmware, But after flashing the new firmware LCD screen is showing Complete even after restart. I am trying to connect via usb no communication.

Marlin fw path: https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin/firmware/Marlin2.0-MKS-Robin

3Dangel, 2019-10-25:
What version of MKS-Robin do you have ?
What version of LCD do you have?
There is a setting to set different version of LCD

github051, 2019-10-25:
Iam using Mks robin 2.4 board and Mks_robin_tft 2.0 lcd

3Dangel, 2019-10-25:
please read through this post

https://github.com/MarlinFirmware/Marlin/issues/9771
cheers

github051, 2019-10-27:
Finaly i'm able to complie and upload the Marlin fw 2.0 in my Mks robin 2.4.
Now i am testing all usefull options one by one.

Thanks to all.

closed 2019-10-27 by github051

daniyo27, 2020-01-14:
@github051 how did you fix this? I am having the same problem with the same board!

3Dangel, 2020-01-14:
Pls download from Marlin official branch and use Atom platformio or Vscode
platformio to compile.

daniyo27, 2020-01-15:
Thank you so much! I was previously using Atom and was presented with multiple compiling errors. VSCode was a breeze to use and my board works now!

3Dangel, 2020-01-15:
Glad to hear it is working!!
Cheers