makerbase-mks/MKS-Robin #143: Robin Nano EEPROM

inib, 2019-11-02:
The Robin Nano has a 16kbit EEPROM (I2C1) and a 64MBit Flash (on SPI2) built in. Has anyone tried to access those with Marlin?

I tried to just activate I2C_EEPROM but this EEPROM won't work with the current Marlin Code and every attempt to access SPI2 kills the FSMC display for me...

makerbase-mks, 2019-11-04:
Currently, the MKS Robin Nano onboard program, the EEPROM is use externally mounted, and does not use the internal EEPROM, the external EEPROM is connected using I2C;
If Marlin firmware is used, this may require the internal EEPROM of the chip. The SPI2 you operate is a touch that is connected to the SPI FLASH and the touch screen. They need to use the CS pin for chip selection.
MKS Robin Nano hardware reference: https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin%20Nano/hardware.

inib, 2019-11-06:
Thank you for your answer. Unfortunately I don't have enough spare time to do this all by myself.
For the external EEPROM a whole new library implementation would be necessary and hardware SPI isn't yet available for Marlin 2 to get a reasonable solution for both flash and touch at the same time.

Would have been great, if this was already implemented, as the default SD-card EEPROM is not very handy.

inib, 2019-11-19:
We have now a working solution for Marlin 2.0.x via SPI2 on the flash.
https://github.com/inib/Marlin/commit/d7077669860182b96a9bfd31403621f71d5cdc78

closed 2019-11-19 by inib