makerbase-mks/MKS-Robin #144: Error compiling

wedgevic, 2019-11-04:
Hi, I am new to marlin and am trying to change the language from my printer from chinese to english, but when I do a PIO Build, I get this error:

Warning! `env_default` configuration option in section [platformio] is deprecated and will be removed in the next release! Please use `default_envs` instead

Can not remove temporary directory `C:\Users\cunha\Documents\Downloads\Marlin2.0_MKS-Robin_mini25\.pioenvs`. Please remove it manually to avoid build issues

Error: Invalid value: invalid choice: 1. (choose from off, chain, deep, chain+, deep+) for option `lib_ldf_mode` in section [env:STM32F1]

Board name: BOARD_MKS_ROBIN_NANO
Build type: build
Environment used: STM32F1

Is there a solution to this? Thanks for the help!

makerbase-mks, 2019-11-05:
1. Click PIO Build in the lower left corner of ATOM, search for mks_robin_nano in the dialog box, and select PIO Build(mks_robin_nano).
2, you can use the Marlin2.0 compiler tool developed by MKS, the connection is as follows: https://github.com/makerbase-mks/MKS-Marlin2.0-Compile-Tool
This will be very helpful.

makerbase-mks, 2019-11-05:
1. Click PIO Build in the lower left corner of ATOM, search for mks_robin_nano in the dialog box, and select PIO Build(mks_robin_nano).
2. You can use the Marlin2.0 compiler tool developed by MKS, the connection is as follows: https://github.com/makerbase-mks/MKS-Marlin2.0-Compile-Tool
This will be very helpful.

wedgevic, 2019-11-05:
I have no option for PIO Build(mks_robin_nano).
I only have these options:
[image]

wedgevic, 2019-11-05:
disregard the name of the project, it is for the robin nano in the config file

makerbase-mks, 2019-11-08:
1.You can change platformio.ini file,like this:
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = mks_robin_nano

2.Compile again

wedgevic, 2019-11-08:
now it gives this: Can not remove temporary directory `C:\Users\cunha\Documents\Downloads\Marlin2.0_MKS-Robin_mini25\.pio\build`. Please remove it manually to avoid build issues

DEPRECATED! A legacy library storage `C:\Users\cunha\Documents\Downloads\Marlin2.0_MKS-Robin_mini25\.piolibdeps` has been found in a project. 

Please declare project dependencies in `platformio.ini` file using `lib_deps` option and remove `C:\Users\cunha\Documents\Downloads\Marlin2.0_MKS-Robin_mini25\.piolibdeps` folder.

More details -> http://docs.platformio.org/page/projectconf/section_env_library.html#lib-deps

Error: Invalid value: invalid choice: 1. (choose from off, chain, deep, chain+, deep+) for option `lib_ldf_mode` in section [env:STM32F1]

Board name: BOARD_MKS_ROBIN_NANO
Build type: build
Environment used: STM32F1

wedgevic, 2019-11-08:
wait, I kinda fixed that, but now there is this error:
.pio\libdeps\mks_robin_nano\TMCStepper_ID5513\src/TMCStepper.h:23:11: fatal error: SoftwareSerial.h: No such file or directory

wedgevic, 2019-11-08:
wait, fixed it!

wedgevic, 2019-11-08:
nope, same error: *** [.pio\build\mks_robin_nano\libc15\TMCStepper_ID5513\source\CHOPCONF.cpp.o] Error 1
In file included from .pio\libdeps\mks_robin_nano\TMCStepper_ID5513\src\source\COOLCONF.cpp:1:0:
.pio\libdeps\mks_robin_nano\TMCStepper_ID5513\src/TMCStepper.h:23:11: fatal error: SoftwareSerial.h: No such file or directory

************************************************************************
* Looking for SoftwareSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SoftwareSerial.h"
* Web  > https://platformio.org/lib/search?query=header:SoftwareSerial.h
*
************************************************************************

  #include <SoftwareSerial.h>
           ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\mks_robin_nano\libc15\TMCStepper_ID5513\source\COOLCONF.cpp.o] Error 1

makerbase-mks, 2019-11-09:
You need to disable TMCStepper@>=0.5.0, <1.0.0, increase at the very beginning
‘#’ symbol

wedgevic, 2019-11-11:
?

sparealcon, 2019-11-29:
I have the same error and I dont understand the solution posted here.

> 
> You need to disable TMCStepper@>=0.5.0, <1.0.0, increase at the very beginning
> ‘#’ symbol

where is this line and in wich file? 
at the beggining of what?
The only difference is I have a mksrobin2  but the error is exactly the same

wedgevic, 2019-11-29:
I did not understand it either

3Dangel, 2019-11-29:
I think it is on the platformio.ini
Where you have to comment out that line.

sparealcon, 2019-12-01:
thanks I was able to fin that line, however it seems the prolem still persists

wedgevic, 2019-12-04:
it didn't work for me either

RavenWarrior, 2020-05-09:
same problem here using Robin_mini with VSCode. exact same problem.
I cant get the original mks robin tft 24 V1.0 to display anything other than the marlin display either.