makerbase-mks/MKS-Robin #145: Where is the smoothieware firmware? For Robin nano

EPDCenter, 2019-11-08:
I want to use Smoothieware, but i dont know where is the firmware

makerbase-mks, 2019-11-09:
1. So far, MKS Robin Nano motherboard can only support its own firmware, and Marlin2.0; does not support Smoothieware firmware;
2. MKS SBASE, MKS SGEN, MKS SGEN_L support Smoothieware firmware and Marlin2.0 firmware, I recommend that if you want to use Smoothieware firmware, I recommend you use MKS SGEN_L motherboard, because it have more inferface and support more LCD，if you use Smoothieware firmware, MKS SGEN_ support MKS LCD12864B and MKS MINI12864…

EPDCenter, 2019-11-09:
Then... why i cant use the robin tft with marlin? where is the mks interface if i use the tft with Marlin?

EPDCenter, 2019-11-09:
What is your firmware? Marlin?

makerbase-mks, 2019-11-15:
MKS designed its own firmware on the Nano board

silentst0rm, 2019-11-20:
> MKS designed its own firmware on the Nano board

This is based on Marlin firmware. You violate the Open Source license without sharing the source code.

EPDCenter, 2019-11-20:
> > MKS designed its own firmware on the Nano board
> 
> This is based on Marlin firmware. You violate the Open Source license without sharing the source code.

+1 @makerbase-mks  está violando la licencia Open Source @silentst0rm Should we publish this on social networks?