makerbase-mks/MKS-Robin #146: Does mks robin nano v1.1 support TMC2209 drivers

CaptainSobeKing, 2019-11-18:
I wanted to confirm these drivers are supported and if the cabling should resemble that of the TMC2208 for UART mode?

makerbase-mks, 2019-11-20:
1. Currently, the MKS Robin Nano motherboard does not have an existing interface using the TMC2208 UART mode.
2, Marlin2.0 should be supported, so you need to use spare IO, as a UART data cable, hardware can refer to: https://github.com/makerbase-mks/MKS-Robin/tree/master/MKS%20Robin %20Nano/hardware

mpalpha, 2020-05-02:
could j18 under the wifi module be used for uart slave addresses for the tmc2209?
what is j18 used for?

mpalpha, 2020-05-27:
Could you please provide a working example using a spare IO as a UART data cable in marlin?

makerbase-mks, 2020-06-15:
@mpalpha This looks great!
@CaptainSobeKing You can refer to @mpalpha tutorial and it is great.