makerbase-mks/MKS-Robin #147: error compile marlin mks robin

NachoFrutos, 2019-11-18:
Hello, I downloaded the marlin 2.0 version of the github of mks robin today. I use to compile Visual Studio Code. But when compiling the default files without modifying anything I get this error.

*** [.pioenvs\mks_robin\firmware.bin] TypeError : a bytes-like object is required, not 'str'
Traceback (most recent call last):
  File "C:\Users\nacho\.platformio\packages\tool-scons\script\..\engine\SCons\Action.py", line 1209, in execute
    result = self.execfunction(target=target, source=rsources, env=env)
  File "C:\Temp\MKS-Robin-master\MKS Robin\firmware\Marlin2.0-MKS-Robin\buildroot\share\PlatformIO\scripts\mks_robin.py", line 25, in encrypt
    robin.write(byte)
TypeError: a bytes-like object is required, not 'str'

makerbase-mks, 2019-11-20:
Can you try to compile with ATOM or use the MKS Marlin 2.0 compilation tool?
https://github.com/makerbase-mks/MKS-Marlin2.0-Compile-Tool

NachoFrutos, 2019-11-20:
Hi, I've tried. because I saw this solution in another post. But I don't get it to work for me. I have tried two computers but I always get an error that I could not download the phyton. but I already have it installed and configured the variables as the manual says.

3Dangel, 2019-11-20:
The latest upgrade from platformio to 4.1 none of the FW can be complied.
Not sure if any other sees this or it is just me.

maureu-too, 2019-11-20:
Unfortunately, after atualization, either vscode, atom or MKS-Marlin2.0-Compile generate the .bin file, but when writing the SD card, my board dont run.

NachoFrutos, 2019-11-20:
Hola, yo he compilador elo firmware con el ATom y me da el mismo error. yo he probado a bajarme la version original del marlin 2.0. Con este codigo si puedo compilar la placa mks robin pero al subir  el firmware se me queda un mensaje en la pantalla tft que pone " COMPLETE" y no hace nada mas.

maureu-too, 2019-11-20:
Hi, I successfully compiled and execute until platformio updated. After that I tried to reinstall on different machines, both windows and linux and it didn't work more. I believe it is some library or something.

3Dangel, 2019-11-20:
Then it is confirmed!!!
Makerbase, would you please test and advise the solution!?
Thanks a lot!!!

makerbase-mks, 2019-11-21:
This may be a bug in Marlin 2.0, and I seem to know where your problem lies.
The Robin.bin file compiled with Marlin 2.0 must have the first character "D8". Otherwise, the update will complete, but the application will not be able to be entered.
MKS re-downloads the latest Marlin, compiles, and produces the .bin file, which is normal for use.
I can't deny that some libraries, or compilers, are flawed, causing the firmware to compile properly, but it can't be used. We will continue to pay attention this problem。

NachoFrutos, 2019-11-22:
Hi, thanks for answering. Well, I'm waiting for news to see if you can solve the problem. I am hopeful because you said you know where it is and you have it identified. Cheers

maureu-too, 2019-11-25:
After several unsuccessful attempts, I downloaded the latest Marlin file again, recompiled without configuring anything. I kept the default setting. After successfully compiling, I made all the necessary changes to MKS-Robin and everything went back to what it was before.
Strangely this happened on two computers after the Platformio upgrade.
I didn't understand what happened, but now it's all right.

makerbase-mks, 2019-11-26:
This may be caused by some library files by the compiler, but it seems difficult to find the ultimate bug. It appears that Marlin has officially resolved this issue.

dnutt20x6, 2019-11-27:
I don't think so, I've been trying to compile for a Robin Mini for the past few days and it fails every time, across 2 different PCs, with the same errors that NachoFrutos posted.

Has anyone else managed to get it to compile in the meantime?

NachoFrutos, 2019-11-29:
Hello, message to makerbase-mks. In this thread they have found a solution. Maybe it can help you find the problem. https://github.com/MarlinFirmware/Marlin/issues/15532

NachoFrutos, 2019-11-29:
good news. with the latest version downloaded today from marlin 2.0. it works now. Please have the manufacturer update the source code.

dnutt20x6, 2019-11-30:
Odd, I redownloaded Marlin 2.0 and it still won't compile for me.

NachoFrutos, 2019-12-03:
Hello, since the latest version of the marlin works the compilation for the mks robin board. We have to use the sample files that come next. Please ask the manufacturer to update the repository with the files that work

dnutt20x6, 2019-12-03:
I confirmed that marlin compiles for the robin mini.

So next I need to use the sample configs to build a proper config then?  All I got was a blank white screen when I flashed the board with the compiled firmware.

makerbase-mks, 2019-12-04:
@dnutt20x6
This is Marlin's bug, you can refer to:https://github.com/makerbase-mks/MKS-Robin-mini/tree/master/marlin%20firmware
or you change pins_MKS_ROBIN_MINI.h :
 #define LCD_RESET_PIN    PF6  --->    #define LCD_RESET_PIN    PC6
I have test it and this is OK.

dnutt20x6, 2019-12-04:
@makerbase-024  Thank you' I've been pulling my hair out all week over this.

The screen works now, but touch is nonresponsive.

dnutt20x6, 2019-12-05:
I did some testing and commenting that line out didn't seem to break anything.

I ran across TH3d's firmware for the Kingroon KP3 and it's more or less identical to the SW200, save that the KP3 lacks a parts cooling fan.

It works perfectly on my printer, so I'm guessing they must have figured out the issue with the touchscreen.  It might be worth a look.

makerbase-mks, 2019-12-06:
@dnutt20x6
Hello, can you provide a link and see the details

dnutt20x6, 2019-12-06:
@makerbase-mks  Sure, here's the page for their firmware:

https://www.th3dstudio.com/knowledgebase/kingroon-kp-3-firmware-marlin-2-0-based/

The source code is provided in the download along with the firmware. and their Github is linked on that page as well.

mattdog01, 2019-12-18:
NachoFrutos
What have you had success with so far?  It seems like you have the following sorted out:
Marlin bug fix 2.0 compiled (was this without errors?)
Your screen is able to display but you don't have the buttons working.
Do you have any details how you got this far?

I have Marlin compiling but with 3 errors
1)…/Marlin-bugfix-2.0.x/Marlin/src/lcd/dogm/status_screen_DOGM.cpp
unused variable 'isHeat' [-Wunused-variable]

2)…/Marlin-bugfix-2.0.x/Marlin/src/lcd/menu/menu_temperature.cpp
array subscript is above array bounds [-Warray-bounds]

3)…/Marlin-bugfix-2.0.x/Marlin/src/lcd/menu/menu_tune.cpp
array subscript is above array bounds [-Warray-bounds]

Using platformio on VS code it will finish compiling. (with the errors above)
I can upload it to the MKS Robin from the SD card.  I get the bar and the percent.
When it is done it displays the "COMPLETED" screen. 
I restart the Robin by cycling the power(I have also tried pressing the reset).
The display shows the "COMPLETED" screen again
The red LED by the part cooling fan connector blinks on.
The fan does the short spin up.
The LED goes off.
Then the Blue LED by the reset button starts blinking.
and that is it :( 
so sad.

Does anyone have any ideas what to try next?

dnutt20x6, 2019-12-29:
@mattdog01  There was some update to VS Code and some of the editor binaries that happened awhile back, and that seemed to fix my compiling errors.

I'd say try using the TH3d firmware I linked above.  You may have to search for your specific printer or a clone of it that uses the same board you have, but it works.

My compiled bin file before I started using ABL was about 211kb for reference, so yeah, yours sounds a bit small.

NachoFrutos, 2019-12-30:
mattdog01 

Hi, I have the plate and tft fully operational. But I have used the original marlin 2.0 bugfix. As soon as I can I upload the two advanced configuration and configuration files. Because you have to activate an option that you don't know so that the tft works by emulating an lcd.

mattdog01, 2019-12-30:
Thanks NachoFrutos,
I'm looking forward to those config files to see what I'm missing.  I really appreciate the help.  I'm so excited to get Marlin working on my board.  It is disappointing to get so little help from Makerbase so thank you.

mattdog01, 2019-12-30:
@dnutt20x6 
I have a home made coryxy printer so I don't see any TH3d printers or clones that are the same.  I have not been able to figure how to tell which board is in which printer on the TH3d site. I'm using the MKS-Robin v2.3 board with the included TFT v1.1 screen in my printer (pre all the other robin flavours). It is a wonderful board but the flexibility of the firmware is very limiting and documentation is basically nonexistent.  I have had to implement some physical workarounds to allow the use of two extruders with one nozzle but that should all be firmware configurable and with Marlin it is. Thanks for the help though.  Hopefully NachoFrutos has the details I'm missing in the config files.

mattdog01, 2019-12-30:
@maureu-too 
Do you have working Marlin 2.0 firmware on your MKS-Robin?
Are there any tricks that you used to overcome any of your issues?

I have it doing this
Using platformio on VS code it will finish compiling.
I can upload it to the MKS Robin from the SD card. I get the upload bar and the percent.
When it is done it displays the "COMPLETED" screen.
I restart the Robin by cycling the power(I have also tried pressing the reset).
The display shows the "COMPLETED" screen again
The red LED by the part cooling fan connector blinks on.
The fan does the short spin up.
The LED goes off.
Then the Blue LED by the reset button starts blinking.

Did you even have this issue?

mattdog01, 2019-12-30:
@makerbase-mks 
What do you mean by this
The Robin.bin file compiled with Marlin 2.0 must have the first character "D8". Otherwise, the update will complete, but the application will not be able to be entered. Do you mean it must start with Ø?

3Dangel, 2019-12-30:
The repository from MKS is an old work from a gentlemen from Russia I
think.
He was able to make it to work with his own coding.
Now the NEW official Marlin2 is not making the TTF V1.1 to work.

mattdog01, 2019-12-30:
@3Dangel 
Thanks for the reply.  So if I want Marlin 2.0 then I need to get a different display? If I'm doing that then I should likely switch board and all, right?

NachoFrutos, 2019-12-31:
Hi, I have the mks robin board operating with your tft. Using the latest version of marlin (bugfix-2.0x - https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x). You have to use the sample configuration files that are located in this path "Marlin / config / examples / Mks / Robin /"

3Dangel, 2019-12-31:
https://github.com/MarlinFirmware/Marlin/issues/16159
this might solve your issue
it did for mine
[image]
[image]
[image]

eddie-moser, 2020-02-22:
you can fix the error by updating line 25 of the mks_robin.py file to this:

byte = chr(ord(byte) ^ key[position & 31]).encode()

mattdog01, 2020-02-22:
I never did come back and report on this.  I have been running Marlin 2.0 on my Robin with the older screen version (V1.1)for several months now and it has been working great.  It has required some tweaking of the settings (mostly due to more accurate temperature measuring) but I'm getting great prints.  I have been using the latest VSCode and the latest Marlin2.0 bugfix and everything compiles and runs perfectly. Fixes that have been worked on in another string have been implemented into the main Marlin 2.0 and they work perfectly. I think this string should be closed by @NachoFrutos as it has been resolved.  Happy Printing.

mattdog01, 2020-02-22:
Hello Eddie,

Thanks for the email.  I have since resolved the problem or rather someone else has resolved the issue and it is now rolled into the main Marlin 2.0.  My printer has been working great for a while.  Thanks again for the note.  Take care.

Matt

makerbase-mks, 2020-02-25:
@mattdog01 Hi, I have checked the Robin.bin file. The first character must be ‘D8’. The other files can be updated, but the program cannot run on the motherboard.However, the new version of marlin seems to have solved this problem.

Kyosuke33, 2020-06-10:
Hello, just a question, i have a Kingroon KP3 who have a Robin mini and a TFT Robin. I upload Marlin and now i have the TFT who is in Marlin mode with 4 touch button in the bottom. It's not possible to run Marlin and keep the Mks touch screen ?
[image]
[image]

makerbase-mks, 2020-06-15:
Hi! @Kyosuke33 Can you provide your configuration.h and your board's pins.h

Kyosuke33, 2020-06-15:
Yes
[Configuration.h.txt](https://github.com/makerbase-mks/MKS-Robin/files/4780962/Configuration.h.txt)
[pins.txt](https://github.com/makerbase-mks/MKS-Robin/files/4780963/pins.txt)

qulet, 2022-11-06:
CZEŚĆ ! @Kyosuke33 Proszę  prześli plik ZIP z całym Folderem MARLINA i osobno plik kompilacj  Robin_mni.bin.  Mam ROBIN_MNI z MKS ROBIN TFT 24 V1.0
Próbuję od kilku tygodni dokonać udanej kompilacji w VSC  niestety bez efektu. Pytanie:  Jak już teoretycznie  będę miał od Ciebie plik ROBIN_MNI.bin  czy wystarcz włożyć go do gniazda karty SD i dokona się aktualizacja.?
Może możesz to udostępnić na dysku GOOGLE ? lub poproszę na maila:
quletqwert@gmail.com
Zgóry dziękuję za życzliwość i poświęcony czas !

qulet, 2022-11-13:
Hello !
I am compiling Marlin for the printer:
TH3D_Unified2_Kingroon_Robin_Mini
receives the Robin_mini.bin file
I check in the HxD program at the beginning there is a D8 bit
Unfortunately, after updating the firmware, the message "Complete" appears
and the Marlin screen does not appear......
What else can I do?

qulet, 2022-11-13:
my bin file

[Robin_mini.zip](https://github.com/makerbase-mks/MKS-Robin/files/9997984/Robin_mini.zip)

Startbh, 2023-10-14:
Hi I need help. I have Easythreed X7 3d printer, and it is working on MKS Robin mini V2. It worked perfectly but last day reported on display an error when he start to print. What can be?
You can start extruder an it is OK, you can move in x,y,z.