makerbase-mks/MKS-Robin #149: Input and output voltage issue

Drakstein, 2019-11-22:
I have a simple question about the voltage of the MKS ROBIN v2.4 board, it suports input 24V, but it also outputs 24V to the motors??? By the way, I will use a BL touch, anyone here already used it, any suggestions? I will use in a core XY model.

makerbase-mks, 2019-11-25:
On the MKS Robin motherboard, the motor drive voltage depends on your power input voltage.

mattdog01, 2019-12-30:
@Drakstein 
The voltage to the stepper motors doesn't seem to really matter because your stepper drivers will only send the set/adjusted current to them.  I run my board on 24volts with a corexy setup using 12volt motors (I think), but I set the current (plus a little bit) to the point where there wasn't missed steps and everything works great.