makerbase-mks/MKS-Robin #155: Firmware not updated anymore via SD Card

jkoenig72, 2019-12-14:
Hi,

some help requested:

I try tui use Marlin 2.x with Tevo Nereus (MKS Nano). Sadly I had some shortcut/smoke on X-Axis stepper driver a4988. No Idea if the nano board survived this.

It boots up, however, some of the graphics on the display are broken and both X and Y axis not working.

Also I can not update the firmware via putting a new .bin file on the SD card.

Any hint - Can I flash the board directly somehow? Or it has to go to the trashbin...?

Any hint - appreciated. 

Thanks!
Joerg

jkoenig72, 2019-12-15:
Maybe there is another way to flas instead sd card?

makerbase-mks, 2019-12-20:
You must copy Robin_nano.bin file to TF card，inset motherboard and repwer.