makerbase-mks/MKS-Robin #160: Error compil mks_robin

vinc40530, 2020-01-18:
good evening it's been 4 days now that I try to compile the marlin but each time I have a compile error telling me that the environment is not found in the boards folder can you help me please

ps: my card is a mks_robin 2.4 and tft 2.0

3Dangel, 2020-01-19:
Platformio.ini
Needs to assign your board as well
Pls check YT for mks-sbase how to Marlin 2 and use that as your guidance.

vinc40530, 2020-01-19:
I have the mks of put in environment by default in plateformio.ini but the json file is missing in the board folder is it normal?

3Dangel, 2020-01-19:
Download from Marlin’s github
Use the config. From there
Now it is at separated branch that you need to download it separately
And use Vscode- platformio to compile

vinc40530, 2020-01-19:
I just did it, it compiles but error again

In file included from C:\Users\pc\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/pinmap.h:22:0,
                 from C:\Users\pc\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PeripheralPins.h:34,
                 from C:\Users\pc\.platformio\packages\framework-arduinoststm32\variants\MKS_Robin2/variant.h:37,
                 from C:\Users\pc\.platformio\packages\framework-arduinoststm32\system\STM32F4xx/stm32f4xx_hal_conf.h:4, 
                 from C:\Users\pc\.platformio\packages\framework-arduinoststm32\system\Drivers\STM32F4xx_HAL_Driver\Inc/stm32f4xx_hal.h:30,
                 from C:\Users\pc\.platformio\packages\framework-arduinoststm32\system\Drivers\CMSIS\Device\ST\STM32F4xx\Include/stm32f4xx.h:250,
                 from C:\Users\pc\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\stm32_def.h:34,
                 from C:\Users\pc\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\PortNames.h:33,
                 from C:\Users\pc\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\PortNames.c:30:
C:\Users\pc\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:16:12: error: 'PortA' undeclared here (not in a function)
   PA_0  = (PortA << 4) + 0x00,
            ^~~~~
C:\Users\pc\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:33:12: error: 'PortB' undeclared here (not in a function); did you mean 'PortA'?
   PB_0  = (PortB << 4) + 0x00,
            ^~~~~
            PortA
C:\Users\pc\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:50:12: error: 'PortC' undeclared here (not in a function); did you mean 'PortB'?
   PC_0  = (PortC << 4) + 0x00,
            ^~~~~
            PortB
C:\Users\pc\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:68:12: error: 'PortD' undeclared here (not in a function); did you mean 'PortC'?
   PD_0  = (PortD << 4) + 0x00,
            ^~~~~
            PortC
C:\Users\pc\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32/PinNames.h:86:12: error: 'PortE' undeclared here (not in a function); did you mean 'PortD'?
   PE_0  = (PortE << 4) + 0x00,
            ^~~~~
            PortD
compilation terminated due to -fmax-errors=5.
*** [.pioenvs\mks_robin2\FrameworkArduino\stm32\PortNames.c.o] Error 1
C:\Users\pc\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp: In function 'uint32_t get_adc_internal_channel(PinName)':
C:\Users\pc\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp:242:10: error: 'PADC_TEMP' was not declared in this scope
     case PADC_TEMP:
          ^~~~~~~~~
C:\Users\pc\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp:242:10: note: suggested alternative: 'PADC_BASE'
     case PADC_TEMP:
          ^~~~~~~~~
          PADC_BASE
C:\Users\pc\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp:257:10: error: 'PADC_VREF' was not declared in this scope
     case PADC_VREF:
          ^~~~~~~~~
C:\Users\pc\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp:257:10: note: suggested alternative: 'PADC_BASE'
     case PADC_VREF:
          ^~~~~~~~~
          PADC_BASE
C:\Users\pc\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp:262:10: error: 'PADC_VBAT' was not declared in this scope
     case PADC_VBAT:
          ^~~~~~~~~
C:\Users\pc\.platformio\packages\framework-arduinoststm32\cores\arduino\stm32\analog.cpp:262:10: note: suggested alternative: 'PADC_BASE'
     case PADC_VBAT:
          ^~~~~~~~~
          PADC_BASE
*** [.pioenvs\mks_robin2\FrameworkArduino\stm32\analog.cpp.o] Error 1
============================================== [FAILED] Took 78.34 seconds ==============================================

vinc40530, 2020-01-20:
In file included from .piolibdeps\mks_robin2\TMCStepper_ID5513\src\source\CHOPCONF.cpp:1:0:
.piolibdeps\mks_robin2\TMCStepper_ID5513\src/TMCStepper.h:23:11: fatal error: SoftwareSerial.h: No such file or directory

************************************************************************
* Looking for SoftwareSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SoftwareSerial.h"
* Web  > https://platformio.org/lib/search?query=header:SoftwareSerial.h
*
************************************************************************

  #include <SoftwareSerial.h>
           ^~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from .piolibdeps\mks_robin2\TMCStepper_ID5513\src\source\DRVCTRL.cpp:1:0:
.piolibdeps\mks_robin2\TMCStepper_ID5513\src/TMCStepper.h:23:11: fatal error: SoftwareSerial.h: No such file or directory

************************************************************************
* Looking for SoftwareSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SoftwareSerial.h"
* Web  > https://platformio.org/lib/search?query=header:SoftwareSerial.h
*
************************************************************************

  #include <SoftwareSerial.h>
           ^~~~~~~~~~~~~~~~~~
In file included from .piolibdeps\mks_robin2\TMCStepper_ID5513\src\source\COOLCONF.cpp:1:0:
.piolibdeps\mks_robin2\TMCStepper_ID5513\src/TMCStepper.h:23:11: fatal error: SoftwareSerial.h: No such file or directory

************************************************************************
* Looking for SoftwareSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SoftwareSerial.h"
* Web  > https://platformio.org/lib/search?query=header:SoftwareSerial.h
*
************************************************************************

  #include <SoftwareSerial.h>
           ^~~~~~~~~~~~~~~~~~
compilation terminated.
compilation terminated.
In file included from .piolibdeps\mks_robin2\TMCStepper_ID5513\src\source\DRVCONF.cpp:1:0:
.piolibdeps\mks_robin2\TMCStepper_ID5513\src/TMCStepper.h:23:11: fatal error: SoftwareSerial.h: No such file or directory

************************************************************************
* Looking for SoftwareSerial.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:SoftwareSerial.h"
* Web  > https://platformio.org/lib/search?query=header:SoftwareSerial.h
*
************************************************************************

  #include <SoftwareSerial.h>
           ^~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs\mks_robin2\lib9a3\TMCStepper_ID5513\source\DRVCONF.cpp.o] Error 1
*** [.pioenvs\mks_robin2\lib9a3\TMCStepper_ID5513\source\COOLCONF.cpp.o] Error 1
*** [.pioenvs\mks_robin2\lib9a3\TMCStepper_ID5513\source\DRVCTRL.cpp.o] Error 1
*** [.pioenvs\mks_robin2\lib9a3\TMCStepper_ID5513\source\CHOPCONF.cpp.o] Error 1

3Dangel, 2020-01-20:
My Mks-Robin is V2.1. Prior the Robin2.
If they are the same family , it uses SPI2 to communicate.
This has to be modified under .platformio source .... (don’t remember the
path)

The spi.cpp needs to be modified to SPI (2);

Google “ MKs Robin markin2”
There was a posted issue on GitHub and resolution given by Jmz52 ( I think)
It has the detailed link on how to do it.

3Dangel, 2020-01-20:
Found it

....... “ Update 2: MKS Robin uses SPI2 to access touch screen controller.
You need to update file
.platformio\packages\framework-arduinoststm32\STM32F1\libraries\SPI\src\SPI.cpp
- change SPIClass SPI(1); to SPIClass SPI(2);.......”

vinc40530, 2020-01-20:
ok, already thanks for the time and your help, so i just changed the classspi, i also installed the tmcstepper library on platformio, and similar, the same error comes back :(

3Dangel, 2020-01-21:
I see there were issues from others as well

check
https://github.com/makerbase-mks/MKS-Robin/issues/120

frank-pian, 2020-05-14:
> In file included from .piolibdeps\mks_robin2\TMCStepper_ID5513\src\source\CHOPCONF.cpp:1:0:
> .piolibdeps\mks_robin2\TMCStepper_ID5513\src/TMCStepper.h:23:11: fatal error: SoftwareSerial.h: No such [...]

I change for `platform = systm32@5.4.3`, the same error.