makerbase-mks/MKS-Robin #164: Marlin 2.0 stuck at 100% loading!

yassinelessawy, 2020-02-02:
Hallo Everybody .
I wanted to install marlin 2.0 on my printer board I have a MKS Robin V2.2 , I have downloaded the latest version v 2.0.3 and downloaded the confg files from the examples (the MKS Robin ones) .and replaced the original files by them I didn't change anything in them I just wanted to see if it will work, I opened PlatformIO in VSCode , choosed the processor STM32F103ZE and Imported the main marlin folder, changed the dafault_envs to mks_robin and then build it and it was successful, "all good so far" I took the Robin.bin from the build folder ".pio" to an sd card to flash the new firmware, I put the sd card into the board it started the loading screen and loads to 100% and a blue led start to flash on the board and nothing else happens.!
Please help, I want this to work.
Thanks in advance.

3Dangel, 2020-02-02:
Robin 2.2
You would need to change SPI class under STM32’s folder in . Platformio.
Robin uses SPI (2) to communicate with touch screen.
More over, you need to identify your version of screen.
Search in google “ MKS Robin Marlin”
The GitHub issue that Shows you how , where to configure SPI
And also the recent fixes for screen V 1.1 & 1.2 that uses different
processor.
Good luck

yassinelessawy, 2020-02-03:
thanks for repyling 3Dangel .from what I understood from the GitHub issue I did  the following I changed the SPI CLASS from (1) to (2) it was the last line in the file, C:\Users\Yassin Ahmed.000\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\SPI\src\SPI.cpp .then i downloaded the lcd calibration tool from https://github.com/jmz52/MKS-Robin-toolkit and entered the values the it gave me in the Configuration.h file in the mks TFT V1.1 (Ihave a MKS ROBIN_V1.1R Screen). Then I build and Uploaded it on the SD card .Everything now working except the screen Stuck At " Booting..." but everthing else works fine ,  I even printed a test Cube From the SD Card , I used Pronterface to select the File. Clearly I did  something wrong in the way, but I have No clue what it is !.
Again Thanks in Advance.

3Dangel, 2020-02-03:
I will see tonight when I get home , if I still have the working Marlin
Source.
But at mean time, have you selected the V1.1 calibration set?

yassinelessawy, 2020-02-03:
First of All thanks for your help, Secondly " have you selected the V1.1 calibration set?" If you mean by that uncommenting it's code in the configuration.h file ,then yes I did that. If you mean something else by it then no and I don't know what are you talking about 😅 still a newbie in this stuff.

3Dangel, 2020-02-03:
The touch bottoms need to be enabled. And there are 2 sets of them. One for
V2 screen and one for V1.1
They are different.

3Dangel, 2020-02-03:
The sets of calibrationX & Y

3Dangel, 2020-02-03:
Make sure you use the tool to see what are your calibration #.
And it tells you also what LCD processor is. V1.1 should be ili9328

yassinelessawy, 2020-02-03:
If this is what you mean! Then I already did that.
[image]

yassinelessawy, 2020-02-03:
[image]

3Dangel, 2020-02-04:
Mmmmmmm
I’m afraid that this is an even older lcd cpu.
Mine was 9328 and just in Nov/Dec got it to work.
It is about the initialization of the LCD that are unique of each model.

yassinelessawy, 2020-02-04:
Okay, so this means that I can't get to to work.😢 Until "if" marlin supports it. Can I use something like Ramps LCD 128x64 smart controller with the mks robin board with a custom beark out board or something?. Thanks for your help.( Glass 3 quarters full I got my machine up and running again, the original FW was horrible random stops and unsteady temp reading the causes even the smallest print to fail.)

3Dangel, 2020-02-04:
Maybe to get another mKS LCD touch screen?
I haven’t got home yet.
Will try to find my working Marlin and see if I can upload it to GitHub.
And you can try.

I know the feeling.
I had a board but couldn’t use it.

But I have an earlier version that still I use the MKS’ FW and still going
strong.
V1.04 is the MKS Version in that board I think.

yassinelessawy, 2020-02-04:
I can't get another mks screen. I can't ship any kind of electronics it will get held at the customs, I got my board from someone that smuggled it through the customs thinking it was mks sbase.😂 The only available 3d printer controller is the mega with ramps 1.4 shield and A4988 drivers with LCD and that is it. Every now and then someone smuggles another Chinese board and sells it for 3 or for times the original price,😂 (NOTE THAT I bought my mks robin for the equivalent of 110 usd)

3Dangel, 2020-02-04:
I totally understand. Been there.

3Dangel, 2020-02-04:
Ok.
Try this.
Your LCD is 9235
The only thing you can try is:
In Marlin/scr/lcd/dogm/u8g_dev_tft_320x240_upscale.....cpp

Find
Case 0x9328:
Change it to
Case 0x9325:
And see if you get lucky.
This is the id that the FW reads from the lcd and use it to go and find the
right initialization sequences.
If this fails then we are done.
Since the codes from jmz52 are merged into the Marlin main branch. So if
you downloaded Marlin from the official Marlin then you will find the above
mentioned .cpp

Good luck!!

yassinelessawy, 2020-02-04:
There is some good news and some bad ones ,the good news is that the screen is now working good but as just a screen not a touch screen and that is the bad news.
Thank you 3Dangel for your help and patience.

yassinelessawy, 2020-02-04:
The touch is working but it is a matter of calibration, ( the four buttons registered at right bottom corner of the screen in a 2mm x 6mm area ).seems like the values of the calibration tool was way off which I am not surprised by, Everytime I do the calibration it gives me a different set of numbers and they are just random values .

3Dangel, 2020-02-04:
Not a problem at all
I’m really happy for you!!!!
Please let me know the outcome with the exact value for the calibration.
Because I might gather all this info and submit it to Marlin and see if
they want to add the model of LCD to be supported. ( you could do it as
well!!) so we help others too.
Really happy to help.
Cheers

yassinelessawy, 2020-02-04:
I can't see the sd card from the screen but I can't print from it as I mentioned earlier ,just select print from sd card from pronterface, is there is a workaround that or it is just the way it is because v2.2 have no 
sd detect pin(I have seen your custom code already in the pins_MKS_Robin.h).
You have been a great help 3Dangel.

yassinelessawy, 2020-02-04:
Once I figure out the exact values I will post them here. (I will leave the issue open till I do.)
Of course I want to help others I just don't know how to submit the data to marlin.

3Dangel, 2020-02-04:
It is try on error.
The SD card mainly will be used manually initializing it.
Do SD detect should be -1.
Will check tomorrow how mine was set.
Cheers

3Dangel, 2020-02-05:
here is the section of the pin definition that you can find in the
pins/stm32/mks-robin

#define ENABLE_SPI2

#if ENABLED(SDIO_SUPPORT)
  #define SCK_PIN           PB13 // SPI2
  #define MISO_PIN          PB14 // SPI2
  #define MOSI_PIN          PB15 // SPI2
  #define SS_PIN            -1   // PB12 is X-
  #define SD_DETECT_PIN     P-1 // SD_CD
  #define ONBOARD_SD_CS_PIN PC11
  #define SDSS              PD2
#else
  // SD as custom software SPI (SDIO pins)
  #define SCK_PIN           PC12
  #define MISO_PIN          PC8
  #define MOSI_PIN          PD2
  #define SS_PIN            -1
  #define ONBOARD_SD_CS_PIN PC11
  #define SDSS              PD2
  #define SD_DETECT_PIN     -1
#endif

yassinelessawy, 2020-02-06:
Hi.
Sorry for replying that late I had  some issues with my laptop ,my pins_MKS_Robin.h is configured as same as yours but the " SD as custom software SPI ( SDIO pins) is dimmed like it is inactive, but it is no big deal I can make a push button and assign a different bin to be the SD_Detect_pin.
My real issue is with the touch calibration, the closest I got the bottle sizes were good and acceptable and there  locations were also good except they were mirrored from left to right, so that the back button who's the enter button and and the up button was the down button and vice versa. When I tried to inverse the signs calibration it didn't work at all,I really don't understand what this values means, can you help me understand what they are. so it would be easier for me to calibrate the touchscreen.
Thanks for your patience and your kindness 3Dangel.

3Dangel, 2020-02-07:
Unfortunately, my knowledge on the subject has reached its limit.
Beyond what you just explained to me, I am not familiar at all with the
symptom your screen has.

Wish you luck and hope you can find the solution you need.
Cheers