makerbase-mks/MKS-Robin #165: Robin Nano + TFT32 - Extruder issue or TFT Issue - Two separate issues.

dmerriman11, 2020-02-17:
Hello,

I'm experiencing two separate issues that have been show stoppers thus far. 

Issue 1
When flashing the close source firmware my extruder motor has no torque, it will turn without a load, but as soon as I feed filament, it stops.
I've adjusted the vRef of the 2208 Driver with no luck.

Issue 2
When flashing Marlin 2.0 open source, my TFT32 screen scrolls down automatically and will not stop, no matter what screen I'm on, it continues to scroll down rendering the UI unusable. 

The TFT32 works flawlessly on the close source version, so I know it's not a defective screen.

Thank you in advance for any assistance.

dmerriman11, 2020-02-19:
Update: I found a copy of the TEVO Nereus closed source firmware and that seems to have fix the torque issue, however I'm still having the Marlin 2.0 TFT calibration issue.

makerbase-mks, 2020-02-19:
Hi, MKS has open sourced the Robin Nano firmware, you can refer to: https://github.com/makerbase-mks/MKS-Robin-Nano-Firmware