makerbase-mks/MKS-Robin #166: Getting Robin nano and TFT35 working with stock Marlin 2.0?

PigeonFX, 2020-04-07:
Has there been a wite up for getting TFT35 working with stock opensource Marlin 2.0? I have seen some people working on it here:
https://github.com/makerbase-mks/MKS-Robin/issues/124

But seems it stopped working, and the last post was in 2019. 

Have seen some people who have got it to work with the Sapphire Pro and a fork here:
https://github.com/inib/Marlin/tree/2.0.X-SapphirePro-3.5TFT

I know I could just use that fork and change the config.h to match my printer but would like to understand the changes needed.

vsinkievic, 2020-04-07:
Look at https://github.com/petrzmax/Marlin. It uses newest Marlin version than inib and it is easier to analyze differences with current Marlin. I checked out current Marlin 2.0.5.2 and taken differences from the petrzmax repository (I used Winmerge in directory compare mode). I updated only few files. And it works!

makerbase-mks, 2020-04-08:
Hi! Yes, you can use the above firmware directly, the function is also relatively complete.

themacboy, 2020-06-28:
Excuse me, there is no official update to make my tft35 work on my Robion nano 1.2? Or im doing something wrong?

Compiling the offical mks repo (this oner) my screen dint work.