makerbase-mks/MKS-Robin #172: RepRap MKS V2 Can't connect to cura

ChrisYOLO, 2020-05-07:
So I got my new custom printer today and I plugged it into my computer. I downloaded CH340 drivers and it was detected inside the device manager (but not detected din printers and scanners). But in cura I was not able to connect it in any way. I tried using German ReRap profile still no connection via USB. My friend suggested using it as a Pursha i3 still no connection. The monitor tab was blank white .

3Dangel, 2020-05-08:
I don’t think 3D printers ever gets detected in windows’ printer/scanner.
It is not a “printer”.

Contact your seller of the custom printer and ask how did they connect to
the PC.

Wrong Baud rate Will not allows you to connect to the PC.

Blank screen you need to ask your seller what firmware they installed in
your unit. Or check for connection cables if they are well connected. Or
you might need to reverse the connectors.
Check YT for reprap full graphic  Issue.
Good luck

ChrisYOLO, 2020-05-08:
> I don’t think 3D printers ever gets detected in windows’ printer/scanner. It is not a “printer”. Contact your seller of the custom printer and ask how did they connect to the PC. Wrong Baud rate [...]

I use an MKS Robin v2 and it gets detected as a USB serial ch340. On other slicers I was able to connect can you help me with the baud rate and com ports on cura I have no ideas how to fix this I am a starter

ChrisYOLO, 2020-05-08:
also i get blank whitet screen on cura my printer is fine no problem but i can't get it to connect with cura

3Dangel, 2020-05-08:
Try using 500000 ; 256000; 112500....
Beyond that, you really need to talk to your seller.
As blank screen, you might need to reflash the right FW.

Watch YT and get some knowdlege about what you have and where you want to
go from there.