makerbase-mks/MKS-Robin #173: MKS Robin Mini auto-off not correct working.

kocmoc007, 2020-05-11:
I have a printer based on the MKS Robin Mini v.2.0 board. I connected the button without fixing to the SW connector. Using the button, the printer turns on / off correctly. Automatic shutdown at the end of printing works only when, after starting printing, select Option - More - Auto-Off in the menu. If I add the M81 command at the end of the G-code, the printer stops, the message Closing machine... appears on the display, and the printer is in this state and does not turn off.
Tell me, how can I automate the process of auto power off, so that every time I do not select this function in the menu?
In the configuration settings, I have installed this:
################## Advanced  Settings#############################

>cfg_insert_det_module			0			#Assemble power outage module (1:mks 220det;0:mks pwc)

>cfg_have_ups_device			0			# UPS(1:yes;0:no)

>cfg_print_over_auto_close		1			# auto-off when print finishes(1:yesЈ»0:no)

makerbase-mks, 2020-05-26:
Hi! I need to know your wiring. Can provide？

kocmoc007, 2020-05-26:
I did not understand the question.  I described in detail everything that I do.  When manually selected in the menu, the printer turns off at the end of printing, but doesn’t, at the command of M81, although it should.  How to do it?