makerbase-mks/MKS-Robin #174: MKS Robin won't work with marlin 2

ashavakloo, 2020-05-14:
hello Everybody
I have a MKS Robin V2.3 board and Robin TFT V1.1R(with R61505U controller). I wanted to install Marlin-bugfix-2.0.x on it. but unfortunately when I uploaded it I have a message on the tft screen that says "COMPLETE" and does nothing else.
What should I do with that, Please help.

3Dangel, 2020-05-15:
I don’t think that LCD controller is supported.
Each different controller has its own initializing sequences, hence if it
is not written by a developer, it won’t be able to be used.

ashavakloo, 2020-05-15:
you mean i can't do anything?

3Dangel, 2020-05-15:
You probably can get the MKS’s firmware to work with them.
And also work with octoprint.
Or you could write the needed initializing codes.

ashavakloo, 2020-05-15:
thanks for replying 3Dangel.i am noob in electronics can i replace TFT controller with somethings else like ILI9328?

3Dangel, 2020-05-16:
Not sure if you could find it around.
It is hard to know ahead of time What LCD caroller it comes with.

The Robin boards are limited due to its design.

I would suggest you to buy something else that has EXP connections that
allows you to connect simple LCD such as Reprap Discount full graphic LCD
or the SKR’s dual LCD

ashavakloo, 2020-05-16:
> write the needed initializing codes.

how can i write initializing codes?

ashavakloo, 2020-05-16:
i changed Case 0x9328 to case 0x1505 In Marlin/scr/lcd/dogm/u8g_dev_tft_320x240_upscale.....cpp
the screen is now working but i have two problems:
1:The writings are reverse
2:TFT display is pale

> you could write the needed initializing codes.

3Dangel, 2020-05-16:
Mmmmm
Search the controller’s spect from mfg
Learn programming lenguage
Learn electronic signals
Learn how to oscilloscope
Learn Marlin LCD codings

I think that should allows you to write some codes