makerbase-mks/MKS-Robin #177: Did not compile MKS-Robin firmware in VS-CODE (version 1.45.1) PlatformIO

patelmontu, 2020-06-10:
Error massage
{
	"resource": "/d:/mksrobin/MKS-Robin-master/MKS Robin/firmware/Marlin2.0-MKS-Robin/.vscode/c_cpp_properties.json",
	"owner": "d:\\mksrobin\\MKS-Robin-master\\MKS Robin\\firmware\\Marlin2.0-MKS-Robin",
	"severity": 4,
	"message": "Cannot find \"d:\\mksrobin\\MKS-Robin-master\\MKS Robin\\firmware\\Marlin2.0-MKS-Robin\\include\".",
	"startLineNumber": 65,
	"startColumn": 21,
	"endLineNumber": 65,
	"endColumn": 98
}

makerbase-mks, 2020-06-15:
Usually, compile board firmware, you need set it,like this(Robin as a example):
1. Set your board type and in configuration.h file
2. Set #define SERIAL_PORT -1
3. Build your firmware

patelmontu, 2020-06-27:
I just download the the firmware from here ([https://github.com/makerbase-mks/MKS-Robin])  and extract it,
And open the vscode and install PlatformIo and click on import arduino project and select the board STM32F103ZE and select the folder MKS Robin>firmware>Marlin2.0-MKS-Robin from extracted folder,
and i just click on Build but it shows the error

{
	"resource": "/d:/mksrobin/Marlin_Firmware/.vscode/c_cpp_properties.json",
	"owner": "d:\\mksrobin\\Marlin_Firmware",
	"severity": 4,
	"message": "Cannot find \"d:\\mksrobin\\Marlin_Firmware\\include\".",
	"startLineNumber": 9,
	"startColumn": 17,
	"endLineNumber": 9,
	"endColumn": 54
}  

PFA Screenshot of error massage 
[image]

3Dangel, 2020-06-27:
You should not import the file, instead you need to open the folder where
your FW is located.
Check YT how to Mks Sbase, it is the same way.
Also, the MKS has its own FW compiler that might be easier to use.
Cheers