makerbase-mks/MKS-Robin #178: Robin Mini V2.0 Extruder motor not turning - new issue ?

jaymcd, 2020-06-15:
I have a Robin Mini on a Kingroon KP3. The extruder stopped working after a I had a filament jam. The extruder motor turns on other driver ports ( x /y ) but the E port seems to no longer work. I can't tell if this is an electrical issue ( i.e. E driver burned out). Nothing evident on the board suggesting this. Is there a way to test the E driver other than plugging a motor in? Everything else on the board seems to work. ??

also - the D3 led light is on - stay on as a red light. Is this normal or indicative of a problem.

makerbase-mks, 2020-06-16:
Hi! Test the extruder method, the temperature of the extruder's heating head reaches above 170 degrees. You can control the rotation of the extruder
D3 is 3.3V power led

jaymcd, 2020-06-16:
Thanks. It doesn't turn even with the extruder preheated. No apparent markings on the board that would indicate the Shelley driver is damaged. Would a jam cause a motor driver to burn out?

Is there a way to test signal coming from the driver?

Jay

Jay McDonald
jaymcd@shaw.ca