makerbase-mks/MKS-Robin #180: MKS robin v1.1 freezing at 100% and won't update again

theprintsurgeon, 2020-06-25:
This issue was addressed previously with someone else but now it is happening again. With the latest marlin bug fix downloaded 6/24/20 the board will not update and freezes on 100%. I have the MKS Robin Lite board v1.1 with the MKS robin lite LCD 12864 v1.1 screen.  I have tried all combinations of serial ports and screens. If someone can just provide me with a working project folder for this board screen combo that would be great. 
[config files.zip](https://github.com/makerbase-mks/MKS-Robin/files/4828949/config.files.zip)

theprintsurgeon, 2020-06-25:
Just trying to compile the most stable release with the MKS Robin Lite board selected I get the following errors : 
	"resource": "/c:/Users/TPS/Documents/3D PRINTER FIRMWARE/marlin stable/Marlin/.pio/libdeps/mks_robin_lite/Adafruit BusIO_ID6214/Adafruit_I2CDevice.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no matching function for call to 'TwoWire::requestFrom(uint8_t, uint8_t, uint8_t)'",
	"startLineNumber": 160,
	"startColumn": 79,
	"endLineNumber": 160,
	"endColumn": 79
}

{
	"resource": "/c:/Users/TPS/Documents/3D PRINTER FIRMWARE/marlin stable/Marlin/.pio/libdeps/mks_robin_lite/Adafruit BusIO_ID6214/Adafruit_SPIDevice.cpp",
	"owner": "cpp",
	"severity": 8,
	"message": "no matching function for call to 'SPIClass::transfer(uint8_t*&, size_t&)'",
	"startLineNumber": 99,
	"startColumn": 31,
	"endLineNumber": 99,
	"endColumn": 31
}

rofoto, 2021-08-06:
@theprintsurgeon have you figured out a solution? Also how have you been able to get the board to take the firmware in the first place. I have the same board and cant even get the firmware to start updating over sd card

theprintsurgeon, 2021-08-06:
The time you are going to spend figuring this out cost more than this board. Just buy a new one.

fjbold1, 2022-06-14:
the lcd uses "REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER"

fjbold1, 2022-06-16:
I got it to compile and the screen works but having a couple of other problems, once the firmware loads it says no sd card recognized, I'm working on it long distance.

mudzify, 2025-07-08:
Hi, I have the same issue, was anyone able to fix this. Screen hanging on 100%.