makerbase-mks/MKS-Robin #184: 'X_SLAVE_ADDRESS' was not declared in this scope

eli76z, 2020-08-09:
Hi all,

i'm trying to upgrade my ender-5 with the latest marlin 2.0 version and i can't compile
getting this error over and over again:
In file included from Marlin\src\HAL\HAL_STM32F1\../../inc/MarlinConfig.h:38:0,
                 from Marlin\src\HAL\HAL_STM32F1\Servo.cpp:26:
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:2031:23: error: 'X_SLAVE_ADDRESS' was not declared in this scope
   INVALID_TMC_ADDRESS(X);
                       ^
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:2029:52: note: in definition of macro 'INVALID_TMC_ADDRESS'
 #define INVALID_TMC_ADDRESS(ST) static_assert(0 <= ST##_SLAVE_ADDRESS && ST##_SLAVE_ADDRESS <= 3, "TMC2209 slave address must be 0, 1, 2 or 3")
                                                    ^~
Marlin\src\HAL\HAL_STM32F1\../../inc/SanityCheck.h:2031:23: note: suggested alternative: 'I2C_ADDRESS'
   INVALID_TMC_ADDRESS(X);
                       ^
i compared the new files with the old ones i have and the config is the same.

any help will be much appreciated, thanks eli.

3Dangel, 2020-08-09:
You might want to report it to Marlin folks.

makerbase-mks, 2020-08-11:
Hi! If you use MKS Robin E3-E3D path firmware and we has update to laset version marlin2.0.6.

3Dangel, 2020-08-11:
Nice to know.
Thanks for the update

eli76z, 2020-08-14:
Thanks for the update.
Can't really use the Bin files you are supplying as I have the Robin E3D on an Ender-5 but hope the update is also for the regular 
Marlin directory, will check if it works now, thanks again.