makerbase-mks/MKS-Robin #185: MKS ROBIN2 V1.0 + TFT35V1.0 Compiling files

LordsWars, 2020-08-27:
Hello, I downloaded your files from GitHub for Robin 2 and compiled it as it is - got a bunch of errors, figured it out and added it to the platformio.ini file -> [env: mks_robin2]
platform = ststm32@5.3.0 got the Robin2.bin code. I noticed that the code is 99 kilobytes when the original is 412510 bytes. I poured it into the board via a flash card and as a result I got the inscription "Booting ..." on the screen and that's it ... the board doesn't work further ... what needs to be changed to load the normal version of Marlin 2.0 ???
I bought this board to create a printer that can drill, cut with a laser and print with plastic, and in the current configuration it can only print from two heads, which is not good ...

3Dangel, 2020-08-27:
You would need to identify your LCD model to see if yours is supported.
There might be three model that is accepted by Marlin.
Also, you would need to select the coordinate set up depending on your LCD
model ( chipset)

Good luck

makerbase-mks, 2020-08-28:
I suggest you use Robin2's own firmware for now, marlin does not currently support F4 FSMC.

LordsWars, 2020-08-28:
That is, I am offered to throw out the purchased board due to the fact that it is not supported by Marlin 2.0 or to use it as it is ... getting rid of the G-code to get what I need ... but this is not the right decision .. The code that is uploaded to the MK is protected, there are no source codes, but what is on the site does not correspond to the validity ... when compiled, it is not working. I have a second profession - a programmer and I work with small tasks on microcontrollers - and I am ready to spend some of my time to figure out and run this microcontroller in the mode I need, a 3-in-1 3D printer ... but I just need to understand whether I will be Can I help the manufacturers of this controller with the source code of this controller or do I still have to understand the structure of Marlin 2.0?

LordsWars, 2020-08-28:
> You would need to identify your LCD model to see if yours is supported. There might be three model that is accepted by Marlin. Also, you would need to select the coordinate set up depending on your LCD model ( chipset) Good luck
> […]
You can determine the type of screen controller ... this is not a big problem ... I worked with many screens and I wrote init for different screens from scratch using only a datasheet .... since there is a pinout of the loop and you need to make a temporary adapter board to determine the screen controller ...

3Dangel, 2020-08-28:
If you are familiar to code, then it would be beneficial to the community
of Marlin that if you are able to bring the screen you have to life.
The screen ini codes are located in features, I think.
Also, Robin uses SPI2 to communicate with the LCD screen And you would need
to change manually the code in platformio as well.
Try to find the past issues in Marlin’s GitHub from google “ Robin LCD not
working” you will most likely to find the post from the past who worked on
different LCD’s MCU.
My LCD was V1.0 and was not supported due to its particular initialization.
And a coder from Russia coded and submitted his portion of code and now is
part of Marlin. And nownI have a working Touch LCD ( emulating Marlin’s
menus)

Another coder has made a small FW just to allow the screen to report its
MCU model. Same coder also made a boot loader ( I think) as well.

Wish you luck.

LordsWars, 2020-08-29:
Well, since "makerbase-mks" are silent, then no one will provide the original source. It is sad.
We need to deal with Marlene.
2 questions arise:
1. Which version of Marlin should I use? the one here or from the original Marlene site?
2. Why is the downloaded version from the manufacturer not loaded? is it incorrect compilation or conversion inside the elf-> bin program ??

3Dangel, 2020-08-29:
Use the Marlin’s site
And use bug fixes

LordsWars, 2020-08-29:
[image]
On the way, everything is standard and everything is connected via FSMC and SPI1.
We will download Marlin 2.0 and compile for a start ...

3Dangel, 2020-08-29:
If you really want to code,  I gonna save you couple hours trying to find
where to start.
Marlin\SRC\LCD\dogm\ug8_dev_tft_320X240_upscale_from......

In there are the ini for different TFT MCU:
ili9328
St7789v
Ili9341

Good luck

LordsWars, 2020-08-29:
> Если вы действительно хотите писать код, я сэкономлю вам пару часов, пытаясь найти, с чего начать. Marlin \ SRC \ LCD \ dogm \ ug8_dev_tft_320X240_upscale_from ...... В нем есть ini для разных TFT MCU: ili9328 St7789v Ili9341 Удачи
> […](#)

Thank you, I have already found these inits. I need to figure out the place of inclusion - which of the defaults is responsible for this ... reverse engineering of the program from inits to default. how to turn on FSMC correctly with this init ... in general, drip and drip :)

LordsWars, 2020-08-30:
Well, here's a little digging, the file tft_io.h was found in which there is a check for the HAS_FSMC_TFT definition, which in turn sends us to tft_fsmc.h where the FSMC pin definitions are registered.
Scrolling through all the pin files found:
`
 #if HAS_FSMC_TFT
  #define FSMC_CS_PIN PD7 // NE4
  #define FSMC_RS_PIN PD11 // A0

  #define TOUCH_CS_PIN PA7 // SPI2_NSS
  #define TOUCH_SCK_PIN PB13 // SPI2_SCK
  #define TOUCH_MISO_PIN PB14 // SPI2_MISO
  #define TOUCH_MOSI_PIN PB15 // SPI2_MOSI

  #define LCD_RESET_PIN PC6 // FSMC_RST
  #define LCD_BACKLIGHT_PIN PD13

  #define TFT_RESET_PIN PC6 // FSMC_RST
  #define TFT_BACKLIGHT_PIN PD13

  #define LCD_USE_DMA_FSMC // Use DMA transfers to send data to the TFT
  #define FSMC_CS_PIN PD7
  #define FSMC_RS_PIN PD11
  #define FSMC_DMA_DEV DMA2
  #define FSMC_DMA_CHANNEL DMA_CH5

  #define TOUCH_BUTTONS_HW_SPI
  #define TOUCH_BUTTONS_HW_SPI_DEVICE 2
#endif
  #define TFT_DRIVER ILI9341
  #define TFT_BUFFER_SIZE 14400

  // YV for normal screen mounting
  // # define ILI9341_ORIENTATION ILI9341_MADCTL_MY | ILI9341_MADCTL_MV
  // XV for 180 ° rotated screen mounting
  #define ILI9341_ORIENTATION ILI9341_MADCTL_MX | ILI9341_MADCTL_MV

  #define ILI9341_COLOR_RGB
`
it remains to check on the correspondence of pins and addresses for STM32F407Zxx which is not enough to add :)

 sorry for my english - google translator ...

LordsWars, 2020-09-04:
Continued .....
editing platformio.ini
#
# mks_robin2_407Z
#
[env: MKS_ROBIN2]
platform = $ {common_stm32.platform}
extends = common_stm32
board = MKS_ROBIN2
build_flags = $ {common_stm32.build_flags}
  -DSTM32F4 -DUSB_PRODUCT = \ "STM32F407ZE \"
  -DTARGET_STM32F4 -DVECT_TAB_OFFSET = 0x8000
extra_scripts = $ {common.extra_scripts}
  pre: buildroot / share / PlatformIO / scripts / mks_robin2.py

then change the version of common_stm32.platform from 6.1.0 to 5.3.0
#
# HAL / STM32 Base Environment values
#
[common_stm32]
platform = ststm32@5.3.0

then copy
Marlin-bugfix-2.0.x \ buildroot \ share \ PlatformIO \
to the boards \ file mks_robin2.json
to the variants \ folder \ MKS_Robin2 \
files are taken from the manufacturer :)

then we try to compile :)
and we run into an error:
C: \ users \ lords \ .platformio \ packages \ framework-arduinoststm32 \ cores \ arduino \ stm32 / PinNames.h: 17: 12: error: 'PortA' undeclared here (not in a function)
   PA_0 = (PortA << 4) + 0x00,

LordsWars, 2020-09-04:
Error solution:
rule
platform_packages = framework-arduinoststm32@3.10500.190327
compile ->OK

LordsWars, 2020-09-05:
Something I hurried ... there were even more mistakes ...
Searching in marlin boards with a 407ZE or 407ZG processor .... I found a few trying to compile BOARD_BTT_SKR_PRO_V1_1 almost compiled swears at #error "Please select only one LCD controller option."
already a plus .... Next, you need to try to compile mks_robin_nano35 there is support for FSMC but the processor 103VE ... and then edit

LordsWars, 2020-09-05:
An attempt to compile any project with a 407 processor leads to errors ... most of the errors are configuration files BUT even substituting the necessary files, other errors arise ... the question arises, is it working in general ??? ok let's try 103 with FSMC ...

LordsWars, 2020-10-21:
Hello, I have a question about the pins in the datasheet on the board there are BUSY and INT signals, and there are no connection of these pins from the screen to the microcontroller .... where are these pins connected ???? I tried to physically ring out - these pins do not reach the display connector ...

LordsWars, 2020-11-06:
[image]
[image]
Well, with grief, I started the board. I checked everything works, it remains only to start the USB - FLASH CARD, this function is not implemented at all on Marlin 2 (USB_FS pin PA11-PA12 - CD-CARD)
For spindle functions, I used pins from the XS1 XS2 connectors, namely REV_IO1-dir, REV_IO2 -enable (when a laser is connected), REV_PWM - PWM, REV_ADC - device temperature sensor.
I have a question: how is USB_FS implemented in your program? inside Marlin there is work with an external flash card BUT it is based on the MAX microcircuit .....

jwchen119, 2021-01-16:
> [image]
> [image]
> Well, with grief, I started the board. I checked everything works, it remains only to start the USB - FLASH CARD, this function is not implemented at all on Marlin 2 (USB_FS pin [...]

Hi,

Hope you're doing well and still playing the same printer.
I'm now having the exact same problem of compiling the firmware for the Robin2 on latest Marlin2.0.
Wonder where's the detailed information of modifying the code to have the right setup?

github051, 2021-02-12:
> [image]
> [image]
> Well, with grief, I started the board. I checked everything works, it remains only to start the USB - FLASH CARD, this function is not implemented at all on Marlin 2 (USB_FS pin [...]

Hi, Can you please share your Robin2 entire folder.

LordsWars, 2021-02-15:
Hello, the original version is no longer there, because I tried to integrate work with a USB flash drive into the system and, as a result, abandoned this business ... the printer prints normally both in 3D printer mode and in laser and milling

[Configuration.zip](https://github.com/makerbase-mks/MKS-Robin/files/5982276/Configuration.zip)
[marlin2.zip](https://github.com/makerbase-mks/MKS-Robin/files/5982299/marlin2.zip)
in the attachment the configuration files, in fact, I launched marlin, changing only their second file is marlin itself 2.0

github051, 2021-02-15:
> Hello, the original version is no longer there, because I tried to integrate work with a USB flash drive into the system and, as a result, abandoned this business ... the printer prints normally [...]

HI,

i downloaded the marlin2.zip and compiled but Robin2.bin file is not generating . Please see the image
[image]

jwchen119, 2021-02-16:
> HI,
>
> i downloaded the marlin2.zip and compiled but Robin2.bin file is not generating . Please see the image
> [image]

I had bunch of warrings errors like this
/framework-arduinoststm32\cores\arduino/avr/io.h:699:6: warning: #warning "device type not defined" [-Wcpp]
\framework-arduinoststm32\cores\arduino\WInterrupts.cpp:56:3: error: 'stm32_interrupt_enable' was not declared in this scope
\framework-arduinoststm32\cores\arduino\WInterrupts.cpp:85:3: error: 'stm32_interrupt_disable' was not declared in this scope

LordsWars, 2021-02-24:
Hello, everything is correct, I use an external STM32 ST-LINK Utility bootloader, it needs a * .BIN file, which in your case means the firmware.bin file after compilation. I did not configure it to rename this file to Robin2.bin, but used everything by default :)

LordsWars, 2021-02-24:
[ROBIN2.zip](https://github.com/makerbase-mks/MKS-Robin/files/6033843/ROBIN2.zip)
here is the BIN file that is now uploaded on my controller, it is possible to work as a 3D printer or as a milling cutter - the laser above I described the connection contacts ...

jwchen119, 2021-03-24:
@LordsWars Hi, I've successfully compiled the firmware. But I have the new problem.
After rename the "firmware.bin" to "Robin2.bin" and manually copy the file into SDcard then reboot, the update process is complete. The screen now stuck at showing "Loading...".  I haven't check the printer is online or not, but this shows that the TFT is not usable. Anything I can do to solve this problem?

BTW, have you ever try to compile the code with latest Marlin-bugfix-2.0.x branch?
I saw that there's new option of TFT section  in "Configuration.h" which might support MKS_Robin_TFT35
[image]

LordsWars, 2021-03-24:
The whole point is who is flashing the MK, if you use the firmware via an SD card, you need to recode with shifts, and when flashing through ST-LINK 2.0, no encoding is required :) I paid attention to the fact that in the original files there is a recoding for writing through a SD card but I did not use it ... I did the firmware directly through ST-LINK 2.0

jwchen119, 2021-03-24:
> The whole point is who is flashing the MK, if you use the firmware via an SD card, you need to recode with shifts, and when flashing through ST-LINK 2.0, no encoding is required :) I paid attention [...]

So what you've done is pressing the Upload button in PlatfomIO to update firmware to the robin2?
BTW, can I have any of personal contact such as IM or e-mail, I think it is faster to just ask directly to you :D

LordsWars, 2021-03-24:
after compiling the program, I close the visual studio, run the program from the ST-LINK 2.0 programmer, then load the * .bin into the program and program the MC through the ST-LINK 2.0 programmer

jwchen119, 2021-03-24:
> after compiling the program, I close the visual studio, run the program from the ST-LINK 2.0 programmer, then load the * .bin into the program and program the MC through the ST-LINK 2.0 programmer

Thanks, I got it. If fail or something happened, can I flash back to factory firmware?
This might not necessary, but can I have your E-mail?

LordsWars, 2021-03-24:
vik_va@mail.ru 
the main language of RU and EN only with a translator :) purely technical, visually understandable but not colloquial :) but about returning to the factory version, if there is a programmer, then there are no problems at all :) it is stitched once :)

LordsWars, 2021-03-24:
ST-LINK 2.0 programmer is easy enough with a blue board https://stm32-base.org/boards/STM32F103C8T6-Blue-Pill

LordsWars, 2021-03-24:
https://istarik.ru/blog/stm32/107.html

jwchen119, 2021-03-24:
Cool, I got it.
If I have any question, I will mail you!

titigafr, 2021-11-01:
Hello, I am trying to compile version 2.0.9.2, I have made good progress but I cannot find the solution to this compilation error when I activate #define SDSUPPORT.
Has anyone managed to compile a recent version of marlin?

thx in advance

LordsWars, 2022-02-03:
[firmware.zip](https://github.com/makerbase-mks/MKS-Robin/files/7995820/firmware.zip)

thinkyhead, 2023-05-08:
@LordsWars — A contributor is now attempting to add a build environment for this board to Marlin over at MarlinFirmware/Marlin#25782. Can you confirm that the environment and board variant are correct in that PR?