makerbase-mks/MKS-Robin #186: MKS Robina Nano v1.2 freeze in booting screen

pawelwo, 2020-08-31:
Hello, while installing a new marlin, the electricity went out. now printer freezes at booting screen, also can't reinstall marlin or original firmware. Printer stops at 100% during installation. does anyone have an idea what i can do?

TheCodeExorcist, 2020-09-21:
Hi,
If the printer stopped at 100% during installation BUT the progress bar was not full then a connection error occured during the firmware flashing. You may try with an other SD card (FAT32 formatted) and/or pushing the card into the slot during the update to avoid contact errors.
However, if the printer stopped at 100% during installation AND the progress bar was full it may be caused by a badly encrypted firmware. Try opening the bin file in an hex editor such as HxD, if it does not start with D8 it can't be flashed on the board using the MKS bootloader. Newer marlin releases fixed this issue, download and compile the latest bugfix from the official marlin github page using an adequate configuration and everything should work fine.

makerbase-mks, 2020-09-23:
@pawelwo 
Hi! I think you should describe the problem clearly in Issues, so that we can find the problem more accurately.
@TheCodeExorcist 
Thank you!

Hrumque, 2021-03-29:
When you put ONLY firmware Robin_nano35.bin file on SD-card, without directory `assets` and bmp-bins inside, this will happen.

mannaei77, 2021-04-21:
dear all
i have purched a new mks robin nano v3 with TS 35 AND TMC 2209,and after installing it on my flsun qqs pro and inserted the sd card it made a sound and the screen was blank ,
i had saved the  ROBIN_NANO_V3.bin and the assets file on the sd card ,
i tried different firmware ,and its the same ,
pleas advise what id needed to make it work .

Hrumque, 2021-04-21:
> dear all
> i have purched a new mks robin nano v3 with TS 35 AND TMC 2209,and after installing it on my flsun qqs pro and inserted the sd card it made a sound and the screen was blank ,
> i had saved [...]

If screeen is blank even when powered on with Sd-card with firmware .bin on it, and dont show something like "updating firmware:" and couning to 100% - this is bad, LCD board or mainboard is dead.

closed 2024-08-01 by pawelwo