makerbase-mks/MKS-Robin #189: MKS Robin e3d 1 + ender 3 pro v2 screen troubleshooting

oguzefe, 2020-10-01:
After coneccted all hardware and installed firmaré ender 3 pro v2 screen only beeps from buzzer and shows only blue screen.

3Dangel, 2020-10-01:
you need to search YT LCD fix.
most likely you would need to reverse connectors on Exp1 and Exp2

oguzefe, 2020-10-02:
Thank you for your advice but I dont know how to reverse exp1 and exp2 because there is only one connection on screen board for exp 3. Added some photos 
[image]
[image]
[image]

3Dangel, 2020-10-02:
Then you would need to find WHAT LCD are you using and WHAT pins you are
using to connect the LCD to the main board and enable the right LCD to be
in communication with the Main board.
Try to seek info on the Facebook forum that deal with your printer model &
mfg
You would have better support and ppl who understand what type of HW you
have and guide you accordingly. My cent and good luck

laertandrade, 2021-09-24:
@oguzefe did you solve the problem with your lcd? I have the same problem.