makerbase-mks/MKS-Robin #190: Wifi Module Not Detected

chiayc, 2020-10-09:
I am using Robin Nano v1.2 with firmware 1.03, I recently purchased a MKS Wifi Module and plugged in to my main board, after power up, I tried to configure the wifi through "Wifi" menu, but I only get "Wifi Module is being configured......." , I waited for 15 minutes and nothing happen. Please help, is it a faulty module or there is something wrong in configuration?

makerbase-mks, 2020-10-09:
Can you provide your config?

chiayc, 2020-10-12:
Hi,

Please see my configuration file. Thanks.

Regards,
Chia

> 
> 
> Can you provide your config?

#===================�߼���������=======================================
>cfg_insert_det_module			0			#�����ϵ�����ģ��(1:mks 220det;0:mks pwc)

>cfg_have_ups_device			0			#�����Ƿ��䱸UPS��Դ(1:��;0:��)

>cfg_print_over_auto_close		0			#�����ػ�����(1:������0:�ر�)�������䡰�����ػ�ģ�顱ʹ��

>PRINT_FINISHED_COUNT 			180			#�����ػ���ʱʱ�䣨��λ���룩

>cfg_filament_det0_trigger_level	0			#E0���ϼ����ӿڵĴ�����ƽ����(1:�ߵ�ƽ����;0:�͵�ƽ����)
>cfg_filament_det1_trigger_level	0			#E1���ϼ����ӿڵĴ�����ƽ����(1:�ߵ�ƽ����;0:�͵�ƽ����)

>cfg_filament_load_length		550			#"����"���ϵĳ���
>cfg_filament_load_speed		800			#"����"�����ٶ�����(mm/min)
>cfg_filament_load_limit_temperature	200			#"����"�������������¶�����

>cfg_filament_unload_length		550			#"����"���ϵĳ���
>cfg_filament_unload_speed		800			#"����"�����ٶ�����(mm/min)
>cfg_filament_unload_limit_temperature	200			#"����"�������������¶�����

#=========================== ��ƽ��ť����  =============================

>cfg_leveling_mode			1			#��ƽģʽ���á�1:�Զ���ƽ��0:�ֶ���ƽ��

#=========================== �ֶ���ƽ ==================================

>cfg_point_number			5			#�ֶ���ƽ�ĸ���:(����3,4,5����)

>cfg_point1:50,50
>cfg_point2:180,50
>cfg_point3:180,180						#�ֶ���ƽʱ��5������λ������
>cfg_point4:50,180
>cfg_point5:100,100

#========================== �Զ���ƽ ===================================

>cfg_auto_leveling_cmd:G28;G29;

#===================��������==========================================
>cfg_language_adjust_type		1	#�����л���ʽ����(1:��Ļ��ť�л�����,0:�����ļ�ѡ���л�����).

>cfg_language_type			1	#����(1:��������;2:��������;3:Ӣ��;4:����;5:��������,6:����,7:��������)��
						#������ֻ����cfg_multiple_language����Ϊ0ʱ��Ч.
#==================�Զ��幦�ܰ�ť����=================================
#���ý����Զ��幦��1
>setmenu_func1_display		1		#1:��ʾ��ť��0:����ʾ��ť

#ÿ��ָ�������÷ֺ�";"����
>setmenu_func1:M84;

>moreitem_pic_cnt		0		#"����"��ť����

#��ť1~��ť7����ָ��
#ÿ��ָ�������÷ֺ�";"����
#ÿ����ť����ָ����������С��200�ֽ�
>moreitem_button1_cmd:G28 X0;G28 Y0;G28 Z0;

>moreitem_button2_cmd:G28 X0;G28 Y0;G28 Z0;

[... 243 lines cut ...]

#������Ϣ
>about_type:Robin_nano43
>about_version:V1.0.3
>about_company:Company:Makerbase
>about_email:E-mail:makerbase@qq.com

#��ͣ/�ָ����������ã���λ��mm��
>PAUSE_UNLOAD_LEN   -3
>RESUME_LOAD_LEN    10

#��ͣ/�ָ��������Ӱٷֱȣ�Ϊ0ʱ��������ã���λ��%��
>RESUME_SPEED    80

#============================= Z��̽������ =============================
>BLTOUCH			1	# 0:����BLTOUCH; 1:����BLTOUCH

#����ģʽ����
>cfg_Standby_mode		0			#�Ƿ��������ѹ��ܣ�0���رգ�1��������
>cfg_Standby_gap_time           600                     #���ý�������ʱ�䣨��λ���룩

>PID_ADJUSTMENT			1			#�������濪��PID���ڹ���
>pid_adjustment_cmd:M303 C5 S200;			#PID����ָ��

#end.