makerbase-mks/MKS-Robin #192: MKS Robin Nano Z home direction revesed and not doing well in middle of work!!

ArMaN2design, 2020-11-28:
Hi...
I have robin nano v1.1 whit drv8825 stp drv on it. whit last firmware online extracted working well. when the board was on and every thing was ok suddenly in between two job an issue happened to z direction home command. without restarting or any thing.
when it began moving to home, it move reverse (upward here) for few mm and then stopped! and never go down again!
I changed direction of home in .cfg file to 1 , it work well in its new direction. when put it to -1 again ,issue was there again.
I change firmware many times, Even install marlin but issue was there, just like first time. changed stepper drive, try to control the board whit pronterface. but changed any thing/

can anyone help me fix this issue pleas?

(sorry about my bad english) :(

3Dangel, 2020-11-28:
Most likely your Z home sensor got stuck

mks-viva, 2020-11-30:
@ArMaN2design 
Hi! You can refer to 3Dangel's idea and check Z sensor. But I hope you can provide a video for me.

ArMaN2design, 2020-12-12:
thanks very much. the problem was exactly from z endstop. the wire was broken from end of one pin. it s work fine now. thanks :)

mks-viva, 2020-12-14:
This is good news, then I will temporarily close this issue.

closed 2020-12-14 by mks-viva