makerbase-mks/MKS-Robin #193: stm32f103zet6  bootloder  where  get?

xin1524, 2020-12-21:
stm32 use isp or serial download program,no bootloder ，bin file can not run，if bootloder can offer?

mks-viva, 2020-12-21:
What motherboard do you use?

xin1524, 2020-12-21:
bootloder for stm32f103 series, maybe stm32f103zet6 is mks-robin boards,i just cautious or confirm if the bootloder is special and must exit flash at the start program.i think all stm32f103 series bootloder is same. if i can write the bootloder or it is not open.

github051, 2021-02-12:
> bootloder for stm32f103 series, maybe stm32f103zet6 is mks-robin boards,i just cautious or confirm if the bootloder is special and must exit flash at the start program.i think all stm32f103 series bootloder is same. if i can write the bootloder or it is not open.

[Uploading MKS_Robin_v2_4_bootloader.zip…]()