makerbase-mks/MKS-Robin #196: Robin2 Lack Of Main Sources

fahrettinn, 2021-03-13:
I bought a MKS Robin2 its has so good properties but it s hard to reach any source there are many problem ;

1-When compiled giving error for Motor Drivers, do not accept any driver.
2-I need to decribe ı/o of drivers but could not pin out diagram and there is no any description for Robin2 about pinout on marlin software.
3-There is no any video about Robin2 on MKS official account.
4-Some source of Robin2 on github not downloadable.

Please helps us MKS, ı m waiting for your feedback :)

3Dangel, 2021-03-15:
Have you tried to contact them via FB? They are much more responsive in
there.
Good luck

fahrettinn, 2021-03-20:
> Have you tried to contact them via FB? They are much more responsive in there. Good luck
> […](#)

What is the via FB?

3Dangel, 2021-03-20:
Facebook
Makerbase MKS has Facebook group. From there, they are more active.

fahrettinn, 2021-03-21:
Thank you so much, I wish solve the problem.

closed 2021-03-21 by fahrettinn