makerbase-mks/MKS-Robin #202: MKS Robin Nano v.3 + TS35 on Delta.

3dSpace116, 2021-06-05:
Hello!
Sorry for my English.
I have MKS Robin Nano v.3 + TS35 and Delta Kossel Linear Plus.
I take marlin 2.0.8 and config files. 
In Platformio.ini I changed default_envs = mks_robin_nano_v3_usb_flash_drive_msc
In Config i set MOTHERBOARD BOARD_MKS_ROBIN_NANO_V3
and TS35, touch and LVGL_UI
But it need calibration menu and some errors, that I don't know how to resolve.

In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL.cpp:28:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:917:6: error: #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
  917 |     #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
      |      ^~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1351:6: error: #error "DELTA_CALIBRATION_MENU requires an LCD Controller."
 1351 |     #error "DELTA_CALIBRATION_MENU requires an LCD Controller."
      |      ^~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1647:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."
 1647 |     #error "LCD_BED_LEVELING requires a programmable LCD controller."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\HAL_SPI.cpp:25:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:917:6: error: #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
  917 |     #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
      |      ^~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1351:6: error: #error "DELTA_CALIBRATION_MENU requires an LCD Controller."
 1351 |     #error "DELTA_CALIBRATION_MENU requires an LCD Controller."
      |      ^~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1647:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."
 1647 |     #error "LCD_BED_LEVELING requires a programmable LCD controller."
      |      ^~~~~
In file included from Marlin\src\HAL\STM32\../../inc/MarlinConfig.h:49,
                 from Marlin\src\HAL\STM32\MarlinSerial.cpp:21:
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:917:6: error: #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
  917 |     #error "ADVANCED_PAUSE_FEATURE requires a supported LCD controller (or EMERGENCY_PARSER)."
      |      ^~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1351:6: error: #error "DELTA_CALIBRATION_MENU requires an LCD Controller."
 1351 |     #error "DELTA_CALIBRATION_MENU requires an LCD Controller."
      |      ^~~~~
Marlin\src\HAL\STM32\../../inc/SanityCheck.h:1647:6: error: #error "LCD_BED_LEVELING requires a programmable LCD controller."
 1647 |     #error "LCD_BED_LEVELING requires a programmable LCD controller."
      |      ^~~~~
*** [.pio\build\mks_robin_nano_v3_usb_flash_drive_msc\src\src\HAL\STM32\HAL.cpp.o] Error 1
*** [.pio\build\mks_robin_nano_v3_usb_flash_drive_msc\src\src\HAL\STM32\HAL_SPI.cpp.o] Error 1
*** [.pio\build\mks_robin_nano_v3_usb_flash_drive_msc\src\src\HAL\STM32\MarlinSerial.cpp.o] Error 1

Then I tryed to connect typical reprap LCD, it compilies without errors, but in pins (stm32 f4) is no pin outs for reprap, its not working.
I'm arduino beginner, detailed explanations would be helpful, thanks. 

- [ ]

spikey1973, 2023-04-05:
I have similar issues here,

using a MKS robin nano v3.1 + ts35 also on a DELTA printer, (DIY geeetech g2spro rebuild) 
#error "DELTA_CALIBRATION_MENU requires an LCD Controller."

Tinbum1, 2023-12-25:
ditto on a Two Trees SP5 v3 with TS35-R v2.0 and Nano v3.1