makerbase-mks/MKS-Robin #203: Installing Marlin on robin Lite v1.1

rofoto, 2021-08-02:
I have been looking for any information on how to update to marlin firmware on the robin Lite v1.1 and have not found anything, Asking here to hopefully get some guidance. 

thanks 

我一直在寻找有关如何在 robin Lite v1.1 上更新到 marlin 固件的任何信息，但没有找到任何信息，在这里询问希望得到一些指导。

谢谢

mks-viva, 2021-08-02:
你直接下载最新版本的Marlin就可以。只需要做以下设置：
1. 板卡类型设置为：BOARD_MKS_ROBIN_LITE
2. 屏幕可以设置为：MKS_MINI_12864 或者 REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

closed 2021-08-03 by mks-viva

TDHofstetter, 2023-02-10:
I'm looking for the same information. I can't find any source for the files required in .../Arduino/Hardware to build for the STM32103 in the Robin Lite V1.1 board... and there are only about three people in the whole Internet talking about it.

k2ksk2k, 2024-09-28:
Use lastest Marlin with branch "bugfix-2.1.x". In the config.ini file , change the settings "ini_use_config = all", "motherboard = BOARD_MKS_ROBIN_LITE", more settings follow your printer configurations. In the Configuration.h, uncomment the "//#define SDSUPPORT" and "//#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER". Use Auto Build Marlin, choose "mks_robin_lite_maple" to build the project.
 Finally the mksLite.bin can be upgraded in the TF card and boot successfully, but lost the lite_cfg.txt and wifi support.

TDHofstetter, 2024-09-28:
Too late for me. That was a year and a half ago; the board is long gone.

k2ksk2k, 2024-09-29:
It took another day to compile "mks_robin_lite" instead of "mks_robin_lite_maple".(Maple build environments are deprecated). 
Add several SD card related deine in the file "Marlin/src/pins/stm32f1/pins_MKS_ROBIN_LITE.h". 
+#ifndef SDCARD_CONNECTION
+  #define SDCARD_CONNECTION              ONBOARD
+#endif
+#define ONBOARD_SPI_DEVICE                     2  // Maple
+#define ENABLE_SPI2
+#define NO_SD_HOST_DRIVE
+
 #define SPI_DEVICE                             2  // Maple
 #define SD_SCK_PIN                          PB13
 #define SD_MISO_PIN                         PB14
 #define SD_MOSI_PIN                         PB15
 #define SD_SS_PIN                           PA15
+#define SDSS                                SD_SS_PIN